felis/USB_Host_Shield_2.0 #557: Trying to adapt USB host Shield 2.0 to read and write hexadecimals using serial comunication

robertovaldes, 2020-10-14:
I have the USB host Shield 2.0 I need to send a hexadecimal via the host to another device. I can already do this via my computer with no problem and i bought the host with the purpose of making my arduino a master (like my computer) since i can't send data slave to slave. Currently I am using a serial debug assistant app that you can get in the windows store to send the hexadecimal to my device. The device i am sending the hexadecimal to has an adapter that changes MDB comunication to usb and viceversa. I have already managed to use the USB host Shield 2.0 to recieve data from a mouse and an xbox controller but i can't seem to make it work with my device. Can someone please help me with this i am stuck. Help would be really appreciated.

xxxajk, 2020-10-14:
Right now I am currently overloaded with work, but i can give to you some tips.
We need details to be able to help you.
1: what is the VID:PID of the device
2: chipset data sheets
3: some sort of pseudo code that demonstrates what you are trying to do.

This is for a vending machine, correct?

Lauszus, 2021-01-24:
Closing this due to lack of reply. 

@robertovaldes please provide more information of the device you are trying to use, your code or any other information that might be relevant.

closed 2021-01-24 by Lauszus