felis/USB_Host_Shield_2.0 #562: Fake PS4 controller

Invictun, 2020-11-11:
I have a fake ps4 controller. The pad works exactly the same as the original. 
Unfortunately, when i try to connect it to the host shield via usb, the controller only starts to charge. 

Is it even possible for the fake ps4 pad to work properly with Host Shield?

Lauszus, 2020-11-15:
First enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging, then it will print the VID and PID of your device. Then you simply add it to the list here: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4USB.h#L24-L26 and modify the `PS4USB` class, so it see your controller as a PS4 controller.

closed 2020-11-15 by Lauszus

Invictun, 2020-11-16:
It doesn't work :(

Serial monitor:
```
23:26:51.897 -> PS4 USB Library StartedHU Init
23:26:52.456 -> Addr:01
23:26:52.456 -> NC:01
23:26:52.456 -> Cnf:01
23:26:52.456 -> HU configured

```

I found PID and VID in Device manager.
VID_054C PID_05C4

So the values in PS4USB. h are correct, but the controller still doesn't want to pair.