felis/USB_Host_Shield_2.0 #563: GetDevDescr () fails on some devices.

YuuichiAkagawa, 2020-11-14:
I am trying to support the Novation Circuit. However, faild on getDevDescr().  
This device returns only 8 bytes of Device Descriptor by default address.
[image]

So I tried the Windows style method and it worked.
https://techcommunity.microsoft.com/t5/microsoft-usb-blog/how-does-usb-stack-enumerate-a-device/ba-p/270685

Usb.cpp
https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L718-L719
```
        // Get device descriptor
//        rcode = getDevDescr(0, 0, sizeof (USB_DEVICE_DESCRIPTOR), (uint8_t*)buf);
        rcode = getDevDescr(0, 0, 8, (uint8_t*)buf);
```

usbh_midi.cpp
https://github.com/YuuichiAkagawa/USBH_MIDI/commit/a718bf6a4372baf022c3e708f4fe816df93894fa#

Then
[image]

Lauszus, 2020-11-15:
Very interesting. Have you seen any other devices that has this behavior?

YuuichiAkagawa, 2020-11-16:
I don't know other devices. I encountered it for the first time.

Lauszus, 2020-11-16:
Okay. For now we can just add it to your MIDI drivers. Feel free to send a PR once the code is ready.

Lauszus, 2020-11-23:
Closing, as https://github.com/felis/USB_Host_Shield_2.0/pull/566 has been merged.

closed 2020-11-23 by Lauszus