felis/USB_Host_Shield_2.0 #567: Help port PS5 controller support from SAMD fork

josephduchesne, 2020-11-26:
I recently got PS5 USB controller support working in the USB host shield SAMD fork (including the fancy trigger haptics), and it should also work with the USB host shield, however I don't have one to check.

Here's the PR in question:
https://github.com/gdsports/USB_Host_Library_SAMD/pull/12

I thought I'd post this here in case anyone comes looking for PS5 controller support, and in hopes that someone with the usb host shield and a PS5 controller could try copying in the new files and testing if it works.

Lauszus, 2020-11-26:
@josephduchesne I do not have a PS5 controller yet, but if I get my hands on one, then I'll add support for it.

By looking at the code the output report is identical to the PS4, so should be trivial to add support for it.

josephduchesne, 2020-11-26:
Not quite identical, a few things moved around. The status update is the larger change, but it is more or less the same as the PS4 version.

PR #585 (2021-01-19): Added support for the PS5 controller
Fixes #567
closed 2021-01-19 by Lauszus