felis/USB_Host_Shield_2.0 #568: Ps4 controller does not pair

struner11, 2020-11-27:
I have trouble pairing my ps4 controller with the arduino and I am wondering if I am doing something wrong. I am using the PS4BT example. When I hold down the share and ps button and the controller starts flashing rapidly, it shows this message:

`Unmanaged HCI Event: 2F, data: 01 D8 04 E6 85 53 A4 00 00 08 25 00 63 BC E6 05 03 24 11 00 12 14 09 57 69 72 65 6C 65 73 73 20 43 6F 6E 74 72 6F 6C 6C 65 72 09 10 02 00 4C 05 CC 09 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 65 04 00 01 00 A6 03 01 04 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 61 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

I haven't found anyone with something similar online so here I am.

Below is the full log:

`PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	83
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	01
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:10
HCI version: 06
Please enable discovery of your device
Unmanaged HCI Event: 2F, data: 01 D8 04 E6 85 53 A4 00 00 08 25 00 63 BC E6 05 03 24 11 00 12 14 09 57 69 72 65 6C 65 73 73 20 43 6F 6E 74 72 6F 6C 6C 65 72 09 10 02 00 4C 05 CC 09 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 65 04 00 01 00 A6 03 01 04 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 65 04 00 01 00 61 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

struner11, 2020-12-02:
I decided to look on google for the HCI event code 2F / 0x2F, and found this comment on another repo, listing all the events: https://github.com/secdev/scapy/issues/2581#issuecomment-626207767
> 0x2F - HCI_Extended_Inquiry_Result

I also had a look at the link he added (https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=478726) and found some explanation to what it does:
> 2.46 INSUFFICIENT SECURITY (0x2F)
The Insufficient Security error code indicates that the HCI command or LMP
PDU sent is only possible on an encrypted link.

> 7.5.7 Read Encryption Key Size command
...
When the HCI_Read_Encryption_Key_Size command has completed, an
HCI_Command_Complete event shall be generated.
If the ACL connection associated with the Connection_Handle is not encrypted,
the Controller shall return an HCI_Command_Complete event with the error
code Insufficient Security (0x2F).

> 7.7.38 Extended Inquiry Result event
...
The HCI_Extended_Inquiry_Result event indicates that a BR/EDR Controller
has responded during the current inquiry process with extended inquiry
response data.

But this does not tell me a whole lot since I don't know much about bluetooth but hopefully somebody will be able to help me out.

struner11, 2020-12-09:
I solved it by trying different dongles, and found one that works.

closed 2020-12-09 by struner11

Virusssssss21, 2022-02-01:
@struner11 Sorry for this late comment. Been trying a lot of different dongles and haven't found one that works. Could you perhaps send me a link of where you bought yours. That would be great because i need it for a school project.