felis/USB_Host_Shield_2.0 #569: Library for MediaTek UART/USB Driver

leocrg, 2020-11-30:
Hi!
First of all I would like to congratulate and thank you for your help in the development of the "USB_Host_Shield_2.0" library for arduino. It was an impeccable job that inspired me to develop a similar library for Mediatek's UART/USB converter chipset (VID: 0E8D & PID: 0023).

I'm Trying to use a GPS Tracker with mediatek chipset to comunicate with the USB Host Shield 2.0 and Arduino MEGA.

When I used the example "hub_demo.ino" I received this respose from the USB:

    01
    --
    String Descriptors:
    Manufacturer:		MediaTek Inc
    Product:		Product

    Device descriptor: 
    Descriptor Length:	12
    Descriptor type:	01
    USB version:		0200
    Device class:		EF
    Device Subclass:	02
    Device Protocol:	01
    Max.packet size:	08
    Vendor  ID:		0E8D
    Product ID:		0023
    Revision ID:		0100
    Mfg.string index:	05
    Prod.string index:	06
    Serial number index:	00
    Number of conf.:	01

    Configuration descriptor:
    Total length:		008D
    Num.intf:		04
    Conf.value:		01
    Conf.string:		00
    Attr.:			80
    Max.pwr:		FA
    Unknown descriptor:
    Length:		08
    Type:		0B
    Contents:	00020A0000010904

    Interface descriptor:
    Intf.number:		00
    Alt.:			00
    Endpoints:		02
    Intf. Class:		0A
    Intf. Subclass:		00
    Intf. Protocol:		00
    Intf.string:		01

    Endpoint descriptor:
    Endpoint address:	81
    Attr.:			02
    Max.pkt size:		0040
    Polling interval:	00

    Endpoint descriptor:
    Endpoint address:	01
    Attr.:			02
    Max.pkt size:		0040
    Polling interval:	00

    Interface descriptor:
    Intf.number:		01
    Alt.:			00
    Endpoints:		01
    Intf. Class:		02
    Intf. Subclass:		02
    Intf. Protocol:		01
    Intf.string:		02
    Unknown descriptor:
    Length:		05
    Type:		24
    Contents:	0010010424
    Unknown descriptor:
    Length:		04
    Type:		24
    Contents:	020F0524
    Unknown descriptor:
    Length:		05
    Type:		24
    Contents:	0601000524
    Unknown descriptor:
    Length:		05
    Type:		24
    Contents:	0103000705

    Endpoint descriptor:
    Endpoint address:	84
    Attr.:			03
    Max.pkt size:		0010
    Polling interval:	03
    Unknown descriptor:
    Length:		08
    Type:		0B
    Contents:	02020A0000030904

    Interface descriptor:
    Intf.number:		02
    Alt.:			00
    Endpoints:		02
    Intf. Class:		0A
    Intf. Subclass:		00
    Intf. Protocol:		00
    Intf.string:		03

    Endpoint descriptor:
    Endpoint address:	82
    Attr.:			02
    Max.pkt size:		0040
    Polling interval:	00

    Endpoint descriptor:
    Endpoint address:	02
    Attr.:			02
    Max.pkt size:		0040
    Polling interval:	00

    Interface descriptor:
    Intf.number:		03
    Alt.:			00
    Endpoints:		01
    Intf. Class:		02
    Intf. Subclass:		02
    Intf. Protocol:		01
    Intf.string:		04
    Unknown descriptor:
    Length:		05
    Type:		24
    Contents:	0010010424
    Unknown descriptor:
    Length:		04
    Type:		24
    Contents:	020F0524
    Unknown descriptor:
    Length:		05
    Type:		24
    Contents:	0603020524
    Unknown descriptor:
    Length:		05
    Type:		24
    Contents:	0103020705

    Endpoint descriptor:
    Endpoint address:	83
    Attr.:			03
    Max.pkt size:		0010
    Polling interval:	03

    Addr:1(0.0.1)

My gps module ST210U is a "black box" and I don't have a good documentation, but I found the module linkit ON (https://seeeddoc.github.io/LinkIt_ONE/#Installing_Mediatek_LinkIt_ONE_SDK) that communicate with the same mediatek driver for PC. When I opened the schematic on EAGLE (https://docs.labs.mediatek.com/resource/linkit-one/en/documentation), I saw that the usb interface is connected in MT2502A IC (https://datasheetspdf.com/pdf-file/947458/MediaTek/MT2502A/1).

Another interesting point is that when I connect to the USB port of my PC two COMPORTS appear, The "MTK USB Debug Port" that from what I understand is used to change the FW and a "MTK USB Modem Port" which I use to communicate and send commands by terminal to GPS ST310U.

Useful Link of inkIt ONE Developer’s Guide: https://docs.labs.mediatek.com/resource/linkit-one/en/documentation

I wonder if someone help me develop this library and make it available on this git. I don't know what I need to know about these drivers and how to get them. Whatever you can do to help me, I would appreciate it very much!

Thanks!

xxxajk, 2020-12-01:
*should* be supported on UHS2. Honestly, I've no idea why not, but I'm guessing because UHS2 is device-centric.
Most likely will work on UHS3 because it is interface-centric.
Check if it works fine on UHS3, should work fine. Let me know if it does not on the UHS3 issues.

Here's the breakdown of your debug dump.
```
Device Class 0xEF Miscellaneous Device and Interface
Device Subclass 0x02 Protocol 0x01 Interface Association Descriptor. 
        The usage of this class code triple is defined in the 
        Interface Association Descriptor ECN that is provided on www.usb.org. 
        This class code may only be used in Device Descriptors.

Intf.number:            00
Intf. Class:            0A CDC-ACM Data
Intf. Subclass:         00
Intf. Protocol:         00

Intf.number:            01
Intf. Class:            02 CDC-ACM Control
Intf. Subclass:         02
Intf. Protocol:         01
```

leocrg, 2020-12-02:
Hi xxxajk,

Which example of this library do you suggest I use for my device? Is there any parameter or register that I should modify before?

xpech, 2021-03-23:
Hi,
I have a close request with a mtk GPS, but the USB Host lib send me this  :
` String Descriptors:
 Manufacturer:		MTK
 Product:		GPS Receiver
 
 Device descriptor: 
 Descriptor Length:	12
 Descriptor type:	01
 USB version:		0200
 Device class:		02
 Device Subclass:	00
 Device Protocol:	00
 Max.packet size:	40
 Vendor  ID:		0E8D
 Product ID:		3329
 Revision ID:		0100
 Mfg.string index:	03
 Prod.string index:	04
 Serial number index:	00
 Number of conf.:	01
 
 Configuration descriptor:
 Total length:		0043
 Num.intf:		02
 Conf.value:		01
 Conf.string:		00
 Attr.:			80
 Max.pwr:		FA
 `
But,the device class is different. "Communication device class". Is there a way to initiate a serial communication ? FTDI seems not working.

xxxajk, 2021-03-27:
Communication class can mean many things, so you need to dump more of the descriptors in order to figure it out. 
It could be something that is totally proprietary. In that case, you would have to sniff the device, and hope to sort it out.