felis/USB_Host_Shield_2.0 #572: Help refactoring code to support native bluetooth on ESP32 boards

jeffmikels, 2020-12-02:
This library is exceptional for using bluetooth dongles with the USB shield, but there are other ways to get bluetooth functionality with an Arduino. ESP32 boards have it built in, for example.

The BT HID libraries would be excellent if they could be abstracted away from their dependency on the USB side of things. Is anyone able to help me fork and refactor the BT libraries to make them work without the USB Host Shield?

ddesign, 2020-12-15:
Also something I would be interested in. Have recently bought an ESP32 for it's built in Wifi and BT specifically for a Mod using game controllers. Am not exactly proficient at C yet so wouldn't want to offer my services specifically for this. I have found this: https://gitlab.com/ricardoquesada/bluepad32 which may be helpful for you, but am struggling to apply (compile) it myself. Am waiting on a new delivery of ESP32 boards as I fear I have fried my last one :)

Lauszus, 2021-02-07:
Would be cool. I might do it at some point, but won't promise anything.

vlinder, 2022-08-21:
Was looking for this, but I guess i'll have to use another library for now.

Would it be possible/desirable to add a small shim in between? Perhaps using the BTD as an interface for it? This probably would not need any refactoring. Just an extra implementation of BTD?

Adapted the example with some pseudo-code to explain.

```C
//USB Usb;
//BTD Btd(&Usb);
BTD Btd = ESP32BT(); // something like this?
PS5BT PS5(&Btd, PAIR);
```

I haven't coded C in over a decade. How much of a lift would this be? Possible?

Lauszus, 2022-09-05:
@vlinder yes it's differently possible, but would require quite a bit of work.