felis/USB_Host_Shield_2.0 #573: USB Midi Interface like Edirol UM-1SX or any other USB MIDI device

UgaDaBuga, 2020-12-06:
I am connecting a ready made MIDI USB device like Edirol UM-1SX interface to Arduino's USB shield and using your library to read MIDI from it and turn on lights.
This Edirol is normaly used to connect a computer via USB to midi devices and simplifies my receiver circuit for midi.

Your library is capable of receiving messages I am sending through Midi in the examples i found
but they are not defined as transport to FortySevenEffects/arduino_midi_library 
and the examples assume you convert this raw midi to serial midi and so on, send it elsewhere using pins... etc. 
But all I want is to decode this Midi signal and use it with the library to call note_on /note off handlers.
But the handlers are not called, and I would manualy need to decode whatever USB poll is reading.

YuuichiAkagawa, 2020-12-20:
We have started developing that.

https://github.com/YuuichiAkagawa/Arduino-UHS2MIDI

YuuichiAkagawa, 2021-05-13:
Released. 
You can install from Arduino Library Manager.