felis/USB_Host_Shield_2.0 #575: FTDI chip USB function documentation?

liudr, 2020-12-15:
Hi,
I was just testing the library with an FT232R and the sample USBLoopback code. It works! But, after long searches on the internet, I can't find any documents of how FTDI IC USB protocol works. I've captured USB data from the sample code and Win 10+FTDI, and RPI+FTDI. This, with the FTDI class helps me understand how FTDI USB functions work but is there any official documentation at all? Thanks.

Lauszus, 2021-01-24:
I think your best bet is to check the documentation at the FTDI website and look through the Linux drivers if you want more information: <https://github.com/torvalds/linux/blob/master/drivers/usb/serial/ftdi_sio.c>.

What specific information are you looking for?

Lauszus, 2021-02-14:
Closing this issue due to lack of response. Feel free reopen.

closed 2021-02-14 by Lauszus