felis/USB_Host_Shield_2.0 #576: Fake ps4 controller, USB connect problem

Invictun, 2020-12-15:
I have a fake ps4 controller.
When i try to connect via usb, the controller starts to charge.

I turned on serial debugging.
Serial monitor:
```
23:26:51.897 -> PS4 USB Library StartedHU Init
23:26:52.456 -> Addr:01
23:26:52.456 -> NC:01
23:26:52.456 -> Cnf:01
23:26:52.456 -> HU configured
```
I found PID and VID in Device manager.
VID_054C PID_05C4

So the values in PS4USB.h are correct, but the controller still doesn't want to pair.

maxmacstn, 2021-01-13:
I also got the counterfeit PS4 controller. It works flawlessly with Mac, iOS (iPad) and Windows. 

However, after one day of trial, I couldn't make it work with USB Host shield in both wired and bluetooth method.

Lauszus, 2021-02-07:
I do not own a fake PS4 controller, so it's hard for me to debug the issue...