felis/USB_Host_Shield_2.0 #580: Usb.init() successfully but can't recognize USB devices ?

quangvankts, 2021-01-08:
Hardware setup:
USB Host Shield Mini -> ESP32
SS -> GND
CLK -> GPIO 18
MISO -> GPIO 19
MOSI -> GPIO 23
INT and RST: unconnected

Power: 
3.3V -> ESP32 3.3V out (tested 3.3V with multimeter)
5V (trace cut) -> ESP's VIN (tested 4.6V with multimeter)

When running "UBS_Desc" example:
- The function "Usb.Init()"  always return 0 (which means USB Host Shield is communicating successfully with EPS32 through SPI ?)
- But when USB devices is plugged in (mouse, USB drive, etc), the function "Usb.getUsbTaskState()" returns 0x12 or 0x13 which means "USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE" or "USB_DETACHED_SUBSTATE_ILLEGAL"
- I tried device which runs on it own DC power (Zoom pedal), but same error.

Any help is much appreciated. Thank you

Lauszus, 2021-01-08:
@quangvankts you need to connect `SS` to `GPIO 9` and `INT` to `GPIO 10`. Please see: https://github.com/felis/USB_Host_Shield_2.0#interface-modifications.

quangvankts, 2021-01-08:
It works great now with SS and INT connected. Thanks !

closed 2021-01-08 by quangvankts