felis/USB_Host_Shield_2.0 #581: Bluetooth not working if SerialMonitor is attached

digibum, 2021-01-08:
I have Arduino Pro Mini ATmega328P 3.3V 8MHz and USB Host Shield with latest library 1.4.0.

I have uploaded example PS3BT and run it widauth USB2SERIAL (widauth serial monitor). The bluetooth dongle starts blinking and PS3 controller connects successfully.

Then I attached USB2SERIAL and started serial monitor. Arduino restared and printed "PS3 Bluetooth Library Started" to serial monitor. And thats it. The program constantly loops and never connects to controller. LED on bluetooth dongle never blinks.

What is happening? Can't I use serial monitor while using bluetooth dongle?
Thank you all!

Lauszus, 2021-01-21:
Yes you can use the serial monitor just fine.

Please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging. It will print some usual statement of what is going on.

xxxajk, 2021-01-21:
out of memory?

Lauszus, 2021-01-22:
@xxxajk I highly doubt it. I've tested the PS3BT example a lot and have never run out of memory. In fact only 47 % of the RAM is used on the Uno for that example: https://github.com/felis/USB_Host_Shield_2.0/runs/1730818468?check_suite_focus=true#step:7:151.

Lauszus, 2021-02-14:
Closing this due to lack of response.

closed 2021-02-14 by Lauszus