felis/USB_Host_Shield_2.0 #586: Bidirectional MIDI not working.

jhsa, 2021-01-17:
Hi, I am using an arduino Pro mini Atmega168 3.3V, 8Mhz.
If I flash the USB_Midi_coverter code, all works and I have a signal on the Arduino TX pin when I send a midi command via USB.

If I flash the Bidirectional_Converter code, I do not get a signal on the Arduino TX pin, and the Arduino LED just blinks repeatedly. A short puls followed by a longer pulse.
Any help would be very much appreciated. Thanks

YuuichiAkagawa, 2021-01-17:
It works on my arduino Pro mini Atmega328 3.3V. (with Arduino MIDI Library v5.0.2)

The bidirectional_converter depends on the behavior of the Arduino MIDI Library.
Does Arduino MIDI Library work on your board?

jhsa, 2021-01-17:
Hi, thanks for your reply.
Well, I have also the v.5.0.2 installed.
What connections need to be made between the shield and the arduino. Only The SPI, SS and RST? Do i also need INT? 
I don't have the boards on top of each other.
Thanks a lot for your time.

jhsa, 2021-01-17:
ok, Changed the board to another pro mini with m328 running at 16Mhz but powered by 3.3V, and it seems I have data on my scope. need to find out if the data is good of course :) Need also to try the other direction. Is it ok that we keep this issue open a bit longer? It might help others. There isn't much info about this bidirectional feature :( 
Thank you for your patience.

jhsa, 2021-01-18:
Ok, if I  send midi data into the host, I seem to have good data on the Arduino TX pin.
But if I send MIDI into the Arduino RX pin, I have bad data on the host output. 
I have a good working midi to USB converter cable connected to the host shield.
First picture shows the correct MIDI data for Program Change 17 on Channel 2.
The second picture shows bad data. it seems that there is data missing at the beginning of the data stream.
So, the code seems not to be processing the midi message correctly.

[image]

[image]

YuuichiAkagawa, 2021-01-18:
> Only The SPI, SS and RST? Do i also need INT?

Please connect INT.

My board is works.

PC -(USB Serial )-> pro mini -> USB Host Shield -(USB)-> Roland UM-ONE -(MIDI)-> Roland Quad Capture -(USB)-> PC 
[image]

jhsa, 2021-01-18:
Well, mine does send something as you can see on the pictures above. But the beginning of the data is missing. INT connects to Pin 9, right?

jhsa, 2021-01-18:
Could it be because I am using an arduino pro mini 16Mhz at 3.3V?

YuuichiAkagawa, 2021-01-18:
> INT connects to Pin 9, right?

Yes.

> 16Mhz at 3.3V?

Overclocking. 
See page 2 of the data sheet.
https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega48A-PA-88A-PA-168A-PA-328-P-DS-DS40002061B.pdf
```
Speed Grade:
 0 - 4MHz@1.8 - 5.5V, 0 - 10MHz@2.7 - 5.5.V, 0 - 20MHz @ 4.5 - 5.5V
```

jhsa, 2021-01-18:
Yeah, I know, but I have some projects where that wasn't a problem.
I am just about to replace the m168 on the other board that didn't work, and solder an m328P chip instead. Will also the XTAL to 8MHz. I don't have a 3.3V regulator with the same footprint, but can power it externally anyway. Let's hope this is the problem.

jhsa, 2021-01-18:
So, Atmega 328P, 8Mhz XTAL, same problem :( It is not sending a valid midi message :( The question is why?

jhsa, 2021-01-18:
Ok, it actually seems to work, but I found out that the difference between the two pictures is that I am  sending data on channel 2 and the host sends only channel 1, no matter what channel I send to the RX pin. This is wrong.

jhsa, 2021-01-18:
I think I know where the problem is.. :) Let's see if I can fix it ;)

YuuichiAkagawa, 2021-01-18:
Quick fix
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USBH_MIDI/bidirectional_converter/bidirectional_converter.ino#L89-L91
```
        default :
          msg[1] = MIDI.getData1();
          msg[2] = MIDI.getData2();
```
to
```
        default :
          msg[0] = MIDI.getType() | MIDI.getChannel() -1;
          msg[1] = MIDI.getData1();
          msg[2] = MIDI.getData2();
```

jhsa, 2021-01-18:
That fix is looking good :) Thanks.. Will test more and report. But it seems to be working.

jhsa, 2021-01-18:
I think we have a winner. I guess you could update the example. Thank you so much.

YuuichiAkagawa, 2021-01-18:
Maybe you are the first to use it properly. Thank you for your report.

jhsa, 2021-01-18:
Glad to help.  Here is the story behind me landing here. 
I help developing a DIY (ONLY) project which is a guitar effect switcher controller based on a STM32F103 board (The "Blue Pill"). It does have USB MIDI and can receive / send messages to a computer or tablet which are USB hosts themselves. But not to control some effect modules. With the USB host capability, it will be possible to control this kind of devices. The problem is that at the moment we have to keep both USB connectors as this library doesn't talk to another host, like a PC. Is this correct?
Our project is for those guitar players like myself that cannot afford expensive controllers. So, we build ours :)

Lauszus, 2021-01-19:
@jhsa what you are looking for is a microcontroller that can emulate a USB device. I would recommend using a Teensy: https://www.pjrc.com/teensy/, as it has great support for it.

For instance it's super easy to emulate a MIDI device: https://www.pjrc.com/teensy/td_midi.html.

jhsa, 2021-01-19:
Thanks for that. But I don't have any Teensy board. I do have a few Blue Pill and Black Pill though. And they are very cheap compared to the Teensy. There is an USB class compliant library for the STM32F103, but it doesn't do USB Host. It would be wonderful if this library here worked also on the STM32F103, but it does not compile. So, for my project I will need a Blue Pill and an Arduino Pro Mini. :)

Lauszus, 2021-01-19:
@jhsa it's pretty easy to add support for any STM32 chip. Please see: https://github.com/Lauszus/Nucleo_F446RE_USBHost

jhsa, 2021-01-19:
Easy for you :D :P  
I the Arduino core I am using for the STM32F103 is the older one from Roger Clark (I think). The one that has support for "Maple". As far as I know there isn't a library for USB Midi for the new STM32 arduino core. Shame.

Lauszus, 2021-01-19:
@jhsa I'm not planning on Adding support for the F103 anytime soon, so either you have to live with the Arduino Pro Mini or add support for it yourself :)

jhsa, 2021-01-19:
You got me wrong. :) I wasn't asking. I can live with using the Pro Mini. Of course, if you find time to do the implementation, it would be nice in general. Perhaps not for me anymore, but for others. The F103 boards are really cheap, just over a dollar.
And I do thank you very much for you suggestions.

jhsa, 2021-01-19:
So, main problem is solved and i think we can close this issue. Thanks a lot for your help. ;)

closed 2021-01-19 by jhsa