felis/USB_Host_Shield_2.0 #587: PS4 constantly disconnects with PS4BT

seanboe, 2021-01-18:
The PS4BT example isn't working well for me. The controller is able to connect fine at first, but after a few seconds it'll disconnect and display the following error:
Disconnect Request: Interrupt Channel
Disconnect Request: Control Channel
Disconnect Request: Control Channel
(with a ton more Disconnect Request: Control Channel errors after).

Then it stops doing anything for a while (a minute or so) and finally prints:
HCI Disconnected from Device
Wait For Incoming Connection Request

, which allows me to reconnect the controller and allow the cycle to begin again.

My dongle has been working reliably, especially for ps3 and the shield and board I'm using (mini USB Host Shield with teensy LC) are new and have been working well for usb examples too. Anyone know what's going on?

seanboe, 2021-01-18:
I've noticed something that seems odd. While charging the controller, the connection is very stable, as expected. After letting it charge for ~1 hour, however, trying to reconnected without charging results in the same problem as before.

Lauszus, 2021-01-19:
It sounds like there is something wrong with the battery inside the controller. What does the `PS4.getBatteryLevel()` function return?

seanboe, 2021-01-19:
I tried reproducing yesterday's problem in order to figure out the battery level, but was unsuccessful regardless of whether I was charging the controller or not... now I'm receiving the error at the bottom of this comment. Instead, I ran the PS4USB example to get the battery level, which seems to reveal the problem, because it returned 0. I guess a battery replacement is warranted? This controller is at least 5 years old... 😅

Here's the other error I was getting:

PS4 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 5C:F3:70:9D:84:81
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Remote Name: Wireless Controller
PS4 controller is connecting
Connecting to HID device
Connected to HID device
Received Key Request
Bluetooth pin is set too: 0000
Pairing successful with HID device
Send HID Control Connection Request
Send HID Control Config Request
Set protocol mode: 00
Send HID Interrupt Connection Request
Disconnect Request: Control Channel

Lauszus, 2021-01-19:
Yeah it looks like the PS4 controller is disconnecting. It's fairly easy to replace the battery in the controller: https://www.ifixit.com/Guide/DualShock+4+Battery+Replacement/22074.

I'll close this issue for now. Feel free to reopen it again if the problem persist :)

closed 2021-01-19 by Lauszus

seanboe, 2021-01-29:
Thought I'd post an update... replaced the battery in the controller, `PS4.getBatteryLevel()` began returning 11, and everything started working normally. 
Thanks Lauszus!