felis/USB_Host_Shield_2.0 #588: Usb.Init() and Usb.Task() Halt Program

TimThacker, 2021-01-21:
I have been working on implementing remote control for a toy jeep that has been outfitted for children with motor disabilities. The basic design is such that the throttle and steering have been replaced with a joystick sending signals to an Arduino Uno, which then communicates with a Sabertooth motor driver to move the vehicle.

Upon installing an Arduino Uno x USB Host Shield combination to the vehicle, all functions are rendered moot and the PS4 controller I have been using for remote control will not connect/pair to the Arduino.  The only solution to this problem has been the removal of the Usb.Init() and Usb.Task() functions, as either one will cause the issue. However, upon the removal of these functions I am unable to pair the controller and any efforts towards implementing wireless control are in vain.

I am using the following Bluetooth Dongle: [Dongle](https://www.officesupply.com/technology/networking-cables/wireless-networking/wireless-nics-adapters/startech-mini-bluetooth-adapter-class-wireless-network-adapter/p206839.html?ref=pla&utm_source=google&utm_medium=cpc&adpos=&scid=scplp206839&sc_intid=206839&gclid=CjwKCAiA6aSABhApEiwA6Cbm_106Hah-wPGY6WNs8cILxBYxY9YIbAUxamyvI1F-zaqFvW6w-jEfThoCgjAQAvD_BwE)
I am concerned that perhaps the dongle, as a class 1 BT device, is drawing more power than is being supplied. 

I lack the knowledge to fully understand how the Usb.Init() and Usb.Task() functions operate and if they are causing some type of interference with the code I am attempting to add to.

`// Include libraries for PS4 Bluetooth and USB Host Shield Support
#include <PS4BT.h>
#include <usbhub.h>
#include <SPI.h>
#include <SoftwareSerial.h>

#define ENABLE_UHS_DEBUGGING 1

USB Usb;
//USBHub Hub1(&Usb); // Some dongles have a hub inside
BTD Btd(&Usb); // Create the Bluetooth Dongle instance

// Create the instance of the PS4BT class
// This will start an inquiry and then pair with the PS4 controller - you only have to do this once
// You will need to hold down the PS and Share button at the same time, the PS4 controller will then start to blink rapidly indicating that it is in pairing mode
PS4BT PS4(&Btd, PAIR);

// Arduino pin assignments
#define SERIAL_OUT           11 
#define SERIAL_IN            10

#define BTN_ESTOP             2
#define BTN_FORWARD           4
#define BTN_REVERSE           3
#define BTN_LEFT              6
#define BTN_RIGHT             5

#define POT_GOVERNOR         A0
#define POT_ACCELERATION     A1 

#define POT_TRIM             A2

#define LED_HEARTBEAT        13
#define LED_ESTOP             7

#define RELAY_FUN1            8
#define RELAY_FUN2            9

// The values in this section only can be modified to alter the behavior of the
// individual ride-on toy.

#define USE_ACCELERATION_POT  true  //If there is no installed acceleration potentiometer, set to 'false'

// This value dictates how quickly scootypuff gains speed when responding to
// a button press. The higher the value, the faster the acceleration.
#define BASE_ACCELERATION     1

// This value dictates how quickly scootypuff loses speed when no buttons are
// being pressed. The higher the value, the faster the acceleration.
#define BASE_DECELERATION     2

// Whether the FUN1/2 relays are in use. Set each to 'true' or 'false.'
#define RELAY_FUN1_ENABLED    true
#define RELAY_FUN2_ENABLED    false

// When INVERT is true, the corresponding relay control pin will go LOW when
// fun is to be engaged. When INVERT is false, it will go HIGH instead. If
// you find your relay is energized when it should be at rest (and vice-versa),
// changing this setting should clear that up.

#define RELAY_FUN1_INVERT     true  // The Ywrobot 1 relay is a LOW activated relay thus this setting should be true
[... 543 lines cut ...]
  //Reading information from the d-pad to control vehicle
  //getButtonPress will return true as long as the button is held down
  if (digitalRead(BTN_ESTOP) == LOW){
    do_estop();
  } else if (PS4.getButtonPress(UP)) {
    do_forward();
    engage_fun(BTN_FORWARD);
    Serial.println("D-PAD FORWARD");
  } else if (PS4.getButtonPress(DOWN)) {
    do_reverse();
    engage_fun(BTN_REVERSE);
    Serial.println("D-PAD REVERSE");
  } else if (PS4.getButtonPress(LEFT)) {
    do_left();
    engage_fun(BTN_LEFT);
    Serial.println("D-PAD LEFT");
  } else if (PS4.getButtonPress(RIGHT)) {
    do_right();
    engage_fun(BTN_RIGHT);
    Serial.println("D-PAD RIGHT");
  } else{
    decelerate(&left_motor);
    decelerate(&right_motor);
  }
}`

Lauszus, 2021-01-24:
The problem is that the USB Host Shield is using pin 13, 12, 11 on an Arduino Uno for SPI communication, thus you can not use them for anything else than SPI.

Furthermore it uses pin 10 and 9 for the SS and INT pin respectively. The SS and INT pins can be relocated if needed. Please see: <https://github.com/felis/USB_Host_Shield_2.0/#interface-modifications>.

Also I would recommend you add a variable to `PS4Parser.cpp` with the timestamp of the last received message here: <https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS4Parser.cpp#L104> similar to what is available in the `PS3BT` driver: <https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3BT.cpp#L342>. Then you should check in your loop that the connection is not suddenly lost, as that could cause the vehicle to keep going at the previous speed without the user being able to stop it without pressing on the emergency button. 

Let me know if you need help with implementing such a functionality and I'll be happy to help.

Lauszus, 2021-01-24:
Also please remove the `delay` in your loop, as you need to call `Usb.Task()` as often as possible (most USB devices needs to be read every 1 ms). Please see the following page for inspiration: <https://www.arduino.cc/en/Tutorial/BuiltInExamples/BlinkWithoutDelay>.

Finally please move `Usb.Task()` outside the `if (estopped != true){` statement, so the USB communication keeps working even when the emergency button is pressed.

TimThacker, 2021-01-24:
Lauszus, thank you! I noticed the SS and INT pins and attempted to relocate them, but completely overlooked the USB Host Shield's use of pins 13, 12, and 11.  I found this to be the issue and the design worked as intended for the most part. A few small issues remain as the code I am adding to uses 11 pins, so I may move from the Arduino Uno to the Arduino Mega.

The inputs were noticeably delayed, and I believe removing delay in the loop as you noted will resolve this. I will investigate the reason for which this was added in the first place, and most likely end up removing it. 

I appreciate your thorough response! I will contact you if I find myself in need of assistance implementing a feature to prevent the vehicle from continuing at previous speed upon unintentional disconnection.

closed 2021-01-24 by TimThacker

Lauszus, 2021-01-24:
@TimThacker you're welcome. Good luck with the project :)

TimThacker, 2021-01-26:
@Lauszus I attempted to implement the feature of checking for lost signal based on timestamped communication. Unfortunately, I was unable to do so successfully. Could you aid me in implementing this?

I wrote my wireless controls such that they would be used if the PS4 controller was connected, and otherwise the controls would be read from a joystick with input pins wired directly to the Arduino. Unfortunately, upon an unintended loss of connection this did not function as intended. I presume the timestamped communication method you mentioned would be more effective.

Also, I tested the removed loop delay and the moved Usb.Task() and had a much more refined wireless control of the vehicle. Thank you for those notes!

Lauszus, 2021-01-26:
Please share your latest code and I'll give you some feedback :)

Lauszus, 2021-01-26:
I've just added a function for you to get the latest timestamp: https://github.com/felis/USB_Host_Shield_2.0/commit/b4bbfecd093b0032fa8795369ed914854a8b706e. Something like this should work:

```C++
if (PS4.connected()) {
  if (millis() - PS4.getLastMessageTime() > 1000) {
    Serial.println("Lost connection to PS4 controller");
    // TODO: Call your function to stop the vehicle
  } else {
    // TODO: Read the PS4 controller
  }
}
```

TimThacker, 2021-01-27:
@Lauszus I am excited to implement this feature to my program and test the functionality and possibilities. I cannot thank you enough for your help!