felis/USB_Host_Shield_2.0 #589: Controller stops working periodically while using a second SPI device

MrOobbo, 2021-01-23:
Hello,

I've been struggling for quite some time now to get this to work:

My goal is to connect a XBOX Controller to an Arduino and send it's data to a second Arduino. To achieve that I'm using a USB Host shield (using the MAX3421E chip) and the NRF24L01+ Radio. Both communicate via SPI with the Arduino. Furthermore I'm using this library "USB_Host_Shield_2.0" Version 1.4.0 (https://github.com/felis/USB_Host_Shield_2.0) and the RF24 library Version 1.3.11 (https://github.com/nRF24/RF24) for the radio.
That's my hardware setup (without the Power lines and the level shifter for simplicity sake):

[image]

Here is my code:

```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <XBOXUSB.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
  #include <spi4teensy3.h>
#endif

#define CE_PIN_RADIO  6 //CE : set high to receive wireless data, low to send
#define CSN_PIN_RADIO 7 //Same as SS: SPI enable pin, set low to listen to SPI bus.
//SS Pin USB needs to stay standard (PIN D10) I don't know how to change it
#define SS_PIN_USB 10 //standard SS pin for USB Host Shield (INT is on 9)

const uint64_t pipe = 0x1; // This is the transmit pipeline
int sendData[1];  // One element array holding our random number

RF24 radio(CE_PIN_RADIO, CSN_PIN_RADIO, 26000000); // Activate the Radio 26Mhz SPI (same as USB SPI);

USB Usb;
XBOXUSB Xbox(&Usb);

unsigned long previousMillis = 0;
const long interval = 20;    // interval at which to read (milliseconds)
bool useRadio = false;

void setup()   
{
  pinMode(CE_PIN_RADIO, OUTPUT);
  pinMode(CSN_PIN_RADIO, OUTPUT);
  pinMode(SS_PIN_USB, OUTPUT);

  Serial.begin(115200); //9600 or 115200
 
  #if !defined(__MIPSEL__)
    while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
  #endif
 
  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1); //halt
  }
  Serial.print(F("\r\nXBOX USB Library Started"));

  //Start Radio after USB
  radio.begin();
  radio.openWritingPipe(pipe);
 
}

void loop()   
{
  
  digitalWrite(SS_PIN_USB, LOW);
  Usb.Task();
  unsigned long currentMillis = millis();

  //getButtonPress and radio.write will alternate every 20ms while USB.task() 
  //can fire every loop cycle
  if (!useRadio && (currentMillis - previousMillis >= interval)){ 
    previousMillis = currentMillis;
    if (Xbox.Xbox360Connected) {
        
        Serial.print("L2: ");
        Serial.print(Xbox.getButtonPress(L2));
        Serial.print("\tR2: ");
        sendData[0] = Xbox.getButtonPress(R2);
        Serial.println(sendData[0]);
    }

    useRadio = true; 
  }
  digitalWrite(SS_PIN_USB, HIGH);
  
  delay(1); //higher delays don't seem to help. No delay makes it worth.
 
  if (useRadio && (currentMillis - previousMillis >= interval)){
     digitalWrite(CSN_PIN_RADIO, LOW);
     radio.write( sendData, sizeof(sendData)); //Nach dem Write Befehl wird CSN wieder auf HIGH gesetzt (siehe RF24.cpp Zeile 98)
     digitalWrite(CSN_PIN_RADIO, HIGH);

    useRadio = false;
  }
  
  Serial.println(sendData[0]);
  
}
```

### Now to the Problem:
I can read values from the controller and send them to a second Arduino, **BUT the value of the controller sometimes gets stuck.** Even without pressing anything it returns some value I pressed before. In my example the Xbox.getButtonPress(R2) command returns the same value for a few seconds regarding the actual trigger position (This also happens with other buttons and analog controls). I know for a fact, that the loop is still running.

The devices by them self (just radio or just USB Host) work flawlessly. As soon as I add `radio.write( sendData, sizeof(sendData)); `to the code, the `Xbox.getButtonPress()` gets stuck. Just commenting out this one line fixes the controller (but than I can't send any data).

I really hope you guys can help... I really dove deep into the libraries, changing pins and settings but nothing helped.

Lauszus, 2021-01-24:
Please do not drive the SS pins manually i.e. remove the `SS_PIN_USB` and `CSN_PIN_RADIO` defines, as the libraries will drive them automatically.

Please also verify that the `RF24_SPI_TRANSACTIONS` is defined: <https://github.com/nRF24/RF24/blob/master/RF24_config.h#L193>, as that is needed to enable SPI transactions in the RF24 library, which is needed so it does not conflict with this library.

Also please remove the delay in the code, as you should call `Usb.Task()` as often as possible.

Another possible issue could be that the 3.3V drops, because the radio is drawing more current than the regulator on the Arduino can supply. Try to supply the radio with an external 3.3V supply to see if that fixes the issue.

Please also try to modify the USB mini shield by cutting the VBUS jumper and then connecting 5V to the VBUS pad. Please see: <https://chome.nerpa.tech/usb-host-shield-hardware-manual/> and <https://chome.nerpa.tech/wp/wp-content/uploads/2011/02/uhm11_manual.jpg>.

MrOobbo, 2021-01-25:
Thanks for your response.
I removed the `SS_PIN_USB` define, but had to leave the `CSN_PIN_RADIO` define in, as creating the RF24 Object `radio` requires me to input its SS pin.

I was able to verify that `RF24_SPI_TRANSACTIONS` is defined all the time. Furthermore I removed the delay, modified the VBUS of the Shield and supplied the radio with its own 3.3V source (I also tried differed configs: no common GND with Arduino, 3.3V external for USB Shield and Radio). 

Unfortunately nothing seemed to help. I still get the same issue. Sometimes worse than others.

Interestingly, running the code from above without the Radio 3.3V and GND plugged in, the USB Host works just fine. As soon as I add the Radio to the mix it breaks... 
Either the code ignores the radio stuff once it can't find it or you are on the right track with the power issues.

MrOobbo, 2021-01-26:
I've done some more testing... It also seems like, the Serial monitor "moves slower" once the radio is plugged in.. it's noticeably sluggish with the radio. Slowing the output/ radio.write interval down doesn't help.

Lauszus, 2021-01-26:
I took at the radio code. A normal `radio.write` can block up to 95 ms: https://github.com/nRF24/RF24/blob/master/RF24.cpp#L907, which explains why the USB connection can become unresponsive or sluggish.

Instead look at the interrupt example, as then you will not be blocking your main loop: https://github.com/nRF24/RF24/blob/master/examples/InterruptConfigure/InterruptConfigure.ino.

Lauszus, 2021-01-26:
Also the max SPI frequency for the nRF chip is 10 MHz - see: https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf page 50, so please call the constructor without setting the SPI baudrate and let the library deal with it i.e:

```C++
RF24 radio(CE_PIN_RADIO, CSN_PIN_RADIO);
```

Lauszus, 2021-02-14:
I'll close this issue, as the issue is really that the radio library is blocking the loop.

closed 2021-02-14 by Lauszus