felis/USB_Host_Shield_2.0 #592: FTDI to read serial data from remote drone

marioarrigonineri, 2021-01-24:
Hi,
I need to read data from a telemetry radio receiver via USB (I need to decode mavlink messages to monitor remote drone)
I'm using and ELEGOO UNO R3 with USB Host Shield to connect the receiver.
It seems I can read bytes using FTDI, however the bytes I read are corrupted and cannot be decoded.
I'm using the same receiver with a PC and a Raspberry ad it works propertly.
I'm using a modified version USBFTDILoopback example with a buffer of 3 bytes to read one byte at a time
Usb dara are as follows:

11:33:26.559 -> 01
11:33:26.559 -> --
11:33:26.559 -> String Descriptors:
11:33:26.559 -> Manufacturer:		FTDI
11:33:26.559 -> Product:		FT232R USB UART
11:33:26.559 -> Serial:			AK06EZF0
11:33:26.559 -> 
11:33:26.559 -> Device descriptor: 
11:33:26.559 -> Descriptor Length:	12
11:33:26.559 -> Descriptor type:	01
11:33:26.559 -> USB version:		0200
11:33:26.559 -> Device class:		00
11:33:26.559 -> Device Subclass:	00
11:33:26.559 -> Device Protocol:	00
11:33:26.559 -> Max.packet size:	08
11:33:26.559 -> Vendor  ID:		0403
11:33:26.559 -> Product ID:		6001
11:33:26.559 -> Revision ID:		0600
11:33:26.593 -> Mfg.string index:	01
11:33:26.593 -> Prod.string index:	02
11:33:26.593 -> Serial number index:	03
11:33:26.593 -> Number of conf.:	01
11:33:26.593 -> 
11:33:26.593 -> Configuration descriptor:
11:33:26.593 -> Total length:		0020
11:33:26.593 -> Num.intf:		01
11:33:26.593 -> Conf.value:		01
11:33:26.593 -> Conf.string:		00
11:33:26.593 -> Attr.:			A0
11:33:26.593 -> Max.pwr:		2D
11:33:26.593 -> 
11:33:26.593 -> Interface descriptor:
11:33:26.593 -> Intf.number:		00
11:33:26.593 -> Alt.:			00
11:33:26.593 -> Endpoints:		02
11:33:26.593 -> Intf. Class:		FF
11:33:26.593 -> Intf. Subclass:		FF
11:33:26.593 -> Intf. Protocol:		FF
11:33:26.593 -> Intf.string:		02
11:33:26.593 -> 
11:33:26.593 -> Endpoint descriptor:
11:33:26.627 -> Endpoint address:	81
11:33:26.627 -> Attr.:			02
11:33:26.627 -> Max.pkt size:		0040
11:33:26.627 -> Polling interval:	00
11:33:26.627 -> 
11:33:26.627 -> Endpoint descriptor:
11:33:26.627 -> Endpoint address:	02
11:33:26.627 -> Attr.:			02
11:33:26.627 -> Max.pkt size:		0040
11:33:26.627 -> Polling interval:	00
11:33:26.627 -> 
11:33:26.627 -> Addr:1(0.0.1)

Lauszus, 2021-01-24:
Can you give me a little more detail? 

What do you mean by "corrupted"? Is data missing? Or is the data complete nonsense?

Can you share your code as well?

marioarrigonineri, 2021-01-24:
Hello
here is the code I'm using

-----------------------------
#include <cdcftdi.h>
#include <usbhub.h>
#include <mavlink.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

class FTDIAsync : public FTDIAsyncOper
{
public:
    uint8_t OnInit(FTDI *pftdi);
};

uint8_t FTDIAsync::OnInit(FTDI *pftdi)
{
    uint8_t rcode = 0;

    rcode = pftdi->SetBaudRate(57600);
    rcode = pftdi->SetBaudRate(115200);

    if (rcode)
    {
        ErrorMessage<uint8_t>(PSTR("SetBaudRate"), rcode);
        return rcode;
    }
    rcode = pftdi->SetFlowControl(FTDI_SIO_DISABLE_FLOW_CTRL);

    if (rcode)
        ErrorMessage<uint8_t>(PSTR("SetFlowControl"), rcode);

    return rcode;
}

USB              Usb;
//USBHub         Hub(&Usb);
FTDIAsync        FtdiAsync;
FTDI             Ftdi(&Usb, &FtdiAsync);

void setup()
{
  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
      Serial.println("OSC did not start.");

  delay( 200 );
}

void loop()
{uint8_t rcode = 0;
    Usb.Task();

    if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
    {
        #define SIZE 3

        uint8_t  buf[SIZE];
        for (uint8_t i=0; i<SIZE; i++)
            buf[i] = 0;

        uint16_t rcvd = SIZE;
        rcode = Ftdi.RcvData(&rcvd, buf);

        if (rcode && rcode != hrNAK)
            ErrorMessage<uint8_t>(PSTR("Ret"), rcode);

        if (rcvd > 2){
          mavlink_message_t msg;
          mavlink_status_t status;
          Serial.print(buf[2]);
          Serial.print(",");
          if(mavlink_parse_char(MAVLINK_COMM_0, buf[2], &msg, &status)) 
            Serial.println("OOOK");
        }
        delay(10);
    }
}
-------------------------------

the code never enters "Serial.println("message read");
I downlaoded the bytes and used in a different   scipt.. bytes read by the PC works, bytes read by arduino do not.
So I think there is something wrong in the way I read data from the receiver.

Lauszus, 2021-01-24:
Note that you set the baud rate twice. Do you want to use `57600` or `115200`?

Also try to remove the `delay(10)`, as you should never use any delays, as you need to call `Usb.Task()` at least once every 1 ms.

To me it sounds like you have not used the correct baud rate or that you are not parsing the data correctly.

Start by creating a minimal example, then send a single byte and confirm that you receive the correct byte on the Arduino. If that works, then you can start decoding the MavLINK data.

marioarrigonineri, 2021-01-25:
thanks.. I used both baud rates (one at a time of course).. with the same (wrong) result.
I'll try acquiring data with no delay.
I was afraid I was using a totally wrong way to read serial data from USB, you feedback is very important. thanks

Lauszus, 2021-02-14:
I'll close this. Feel free to reopen if you still think there is an issue with the library.

closed 2021-02-14 by Lauszus