felis/USB_Host_Shield_2.0 #594: Connecting MiniDSP 2x4HD - How to diagnose issue

dennisfrett, 2021-01-24:
I am trying to create an Arduino version of this NodeJS project: https://github.com/mrene/node-minidsp.

It connects to the MiniDSP 2x4HD device over USB and reads out status / sets configuration.

I've set up the USB Host Shield Mini 2.0 to connect to an Arduino Pro Mini 3.3V.

When running the USB HID Description example, everything seems to be working fine for a USB keyboard I'm attaching, but my DSP never gets initialized. I have not cut the VBUS trace to supply 3.3V, but the DSP is not bus-powered, so it probably should not matter. As a test, I've also connected the DSP through a powered hub. Again, the device does not initialize, but the keyboard works fine.

The result I get from the USB Description example is:
<details><summary>Click here to expand output</summary>

```
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		EF
Device Subclass:	02
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		2752
Product ID:		0011
Revision ID:		06F2
Mfg.string index:	01
Prod.string index:	03
Serial number index:	00
Number of conf.:	02
Total length truncated to 256 bytes
Configuration descriptor:
Total length:		0175
Num.intf:		05
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		00
Unknown descriptor:
Length:		08
Type:		0B
Contents:	0003010020000904

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		20
Intf.string:		03
Unknown descriptor:
Length:		09
Type:		24
Contents:	010002087F00000824
Unknown descriptor:
Length:		08
Type:		24
Contents:	0A29030700090824
Unknown descriptor:
Length:		08
Type:		24
Contents:	0B28012903081124
Unknown descriptor:
Length:		11
Type:		24
Contents:	02020101002802000000000B0000061224
Unknown descriptor:
Length:		12
Type:		24
Contents:	060A020F0000000F0000000F000000000C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03140103000A280000001124
Unknown descriptor:
Length:		11
Type:		24
Contents:	02010102002802000000000D0000001A24
Unknown descriptor:
Length:		1A
Type:		24
Contents:	060B010F0000000F0000000F0000000F0000000F000000000C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03160101000B280000070904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		04

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		04
Unknown descriptor:
Length:		10
Type:		24
Contents:	010200010100000002000000000B0624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020103180705

Endpoint descriptor:
Endpoint address:	01
Attr.:			05
Max.pkt size:		0126
Polling interval:	01
Unknown descriptor:
Length:		08
Type:		25
Contents:	0100000208000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			11
Max.pkt size:		0004
Polling interval:	04

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		04
Unknown descriptor:
Length:		10
Type:		24
Contents:	010200010100000002000000000B0624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020102100705

Endpoint descriptor:
Endpoint address:	01
Attr.:			05
Max.pkt size:		00C4
Polling interval:	01
Unknown descriptor:
Length:		08
Type:		25
Contents:	01000108C6000000

Total length truncated to 256 bytes
Configuration descriptor:
Total length:		0175
Num.intf:		05
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		00
Unknown descriptor:
Length:		08
Type:		0B
Contents:	0003010020000904

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		20
Intf.string:		03
Unknown descriptor:
Length:		09
Type:		24
Contents:	010002087F00000824
Unknown descriptor:
Length:		08
Type:		24
Contents:	0A29030700090824
Unknown descriptor:
Length:		08
Type:		24
Contents:	0B28012903081124
Unknown descriptor:
Length:		11
Type:		24
Contents:	02020101002802000000000B0000061224
Unknown descriptor:
Length:		12
Type:		24
Contents:	060A020F0000000F0000000F000000000C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03140103000A280000001124
Unknown descriptor:
Length:		11
Type:		24
Contents:	02010102002802000000000D0000001A24
Unknown descriptor:
Length:		1A
Type:		24
Contents:	060B010F0000000F0000000F0000000F0000000F000000000C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03160101000B280000070904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		04

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		04
Unknown descriptor:
Length:		10
Type:		24
Contents:	010200010100000002000000000B0624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020103180705

Endpoint descriptor:
Endpoint address:	01
Attr.:			05
Max.pkt size:		0126
Polling interval:	01
Unknown descriptor:
Length:		08
Type:		25
Contents:	0100000208000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			11
Max.pkt size:		0004
Polling interval:	04

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		04
Unknown descriptor:
Length:		10
Type:		24
Contents:	010200010100000002000000000B0624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020102100705

Endpoint descriptor:
Endpoint address:	01
Attr.:			05
Max.pkt size:		00C4
Polling interval:	01
Unknown descriptor:
Length:		08
Type:		25
Contents:	01000108C6000100

Addr:1(0.0.1)
```

</details>

So it can actually see the USB device and enumerate it. The PID and VID are correct. I just cannot use the HID library to interact with it. 

When implementing my own sub-class of `HIDUniveral`, `HIDUniversal::isReady()` never returns `true` for the DSP, but it does for the keyboard. 

How should I debug where the problem lies? Looking at the NodeJS version it seems clear to me that this device follows the HID standard. Am I missing something important?

Lauszus, 2021-01-24:
Please start by enable serial debugging if you haven't already: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging.

I've checked the code of the <https://github.com/mrene/node-minidsp> and I see no reason why it shouldn't be able to work with this library.

The [PSBuzz](https://github.com/felis/USB_Host_Shield_2.0/blob/master/PSBuzz.h) driver might be a better starting point, as it shows how to add a custom HID device. Start by modifying the `VID` and `PID`: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PSBuzz.h#L24-L25 to match your device: <https://github.com/mrene/node-minidsp/blob/master/src/constants.js#L3-L4> and then go from there.

When you are done I'll be more than happy to merge the driver into this library, so other people can benefit from your work :)

dennisfrett, 2021-01-24:
Hi, thanks for the quick reply!

The `PSBuzz` driver with a modified PID and VID is what I was trying before.
I've now enabled debugging, and added some custom code to `PSBuzz::connected`. Whenever `HIDUniversal::isReady()` returns true, but the VID / PID check fails, I print something to the terminal.

When plugging in my keyboard in this case, the message from the modified `PSBuzz::connected` call gets printed. So `isReady()` is true, but the VID / PID is wrong, as expected. The following debugging info is then logged:
```
HU Init
Addr:0B
NC:01
Cnf:01
HU configured
```

As an extra test, I plugged in a random USB smart card reader (not sure if it's a HID device), and then this debugging info gets printed:
```
HU Init
Addr:0B
NC:01
```

When I plug in the DSP, nothing gets printed to the terminal at all, not even `HU Init`.

There might be something stupid that I'm missing here. What would the next step in identifying the actual issue be?

I would be glad to provide the driver when it's finished!

dennisfrett, 2021-01-24:
I did some more testing, and it seems like the DSP is only detected the first time I plug it into the shield. After that the DSP works fine, but any USB connection to it (through the normal PC application) just fails. I have to restart the DSP to get it to connect again.

The debug output for the first time is:
```
HU Init
Addr:01
NC:02
Cnf:01
HU configured
```
I then see that it calls `PSBuzz::OnInitSuccessful`. `HIDUniversal::isReady()` never seems to be `true`.

Lauszus, 2021-01-24:
Hmm that does not make sense, as if `HU configured` is printed, then `HIDUniversal::isReady()` will return true. See: https://github.com/felis/USB_Host_Shield_2.0/blob/master/hidcomposite.cpp#L248 and https://github.com/felis/USB_Host_Shield_2.0/blob/master/hidcomposite.h#L100.

I don't really understand if you have modified the PID or VID or not? You need to remove `PSBUZZ_VID` and `PSBUZZ_PID` and replace them with the correct PID and VID. If you don't then the following check will fail: https://github.com/felis/USB_Host_Shield_2.0/blob/master/Usb.cpp#L746 and the device will initialized.

Maybe it's easier if you create a fork and open up  a pull request? Then I can comment directly on your code.

dennisfrett, 2021-01-24:
Okay, I figured out the issue. My powered hub doesn't actually properly work when the input USB voltage as 3.3V. I went the route of cutting the VBUS trace, connecting it to 5V, and not the DSP gets correctly detected.

I'm sorry for bothering you with this. Once the driver is working, I'll create a pull request for it!

Lauszus, 2021-01-25:
Good to hear that you got it working! Looking forward to the PR :)

dennisfrett, 2021-01-25:
I'm still having some issues interfacing with the device. 

To verify, I'm trying to run the USB HID Descriptor example, unmodified (Debugging not enabled at the moment).

I get back: `5`.
It seems that this is the code for `STALL`.

Is it expected that any HID-conforming device should correct work with this example, or could this be due to lack of a driver?
It seems to initialize / configure correctly, and then fail at `FailGetReportDescr1`.

If needed I can re-run the example later with debugging enabled again.

Lauszus, 2021-01-26:
Can you read the HID descriptor in Linux or similar?

The issue could be due to the fact that the total length of the descriptor is hardcoded to be 128 bytes: https://github.com/felis/USB_Host_Shield_2.0/blob/cd87628af4a693eeafe1bf04486cf86ba01d29b8/usbhid.cpp#L39

You can get the size by reading the configuration descriptor - see: <https://stackoverflow.com/questions/52106252/how-to-request-usb-hid-report-descriptor-from-hid-devices-using-usbdeviceconnect>.

Why do you want to read the HID descriptor in the first place? You do not need that, as you can always do that on your computer or in this case you can simply just port the NodeJS driver.

dennisfrett, 2021-01-27:
I was under the impression I should get at least the HID Descriptor example to work before I could continue. That was not correct it seems.

I will check later if I can read it on Linux.

The reason I was trying this, was because I was having trouble porting the driver. The DSP never responded. I am quite new to USB HID, so I was unsure which USB Host Shield 2.0 APIs map to the node-hid APIs.

node-hid has: `write` and `sendFeatureReport`. The node-minidsp project uses `write`.
Does this map to `outTransfer` or `ctrlReq` in this library? I tried both, but got no response from the device yet.

When looking at the packets that the official MiniDSP clients sends in Wireshark, they are: `URB_INTERRUPT out`.

Lauszus, 2021-01-31:
I would say that it would take you longer to read and understand the HID descriptor (if you are not already familiar with it) than just starting writing the driver directly :)

Based on the `write` function here: https://github.com/mrene/node-minidsp/blob/master/src/transport/usb.js#L49-L67 I believe your code should look something like this:

```C++
uint8_t buf[65];
memset(buf, 0xFF, sizeof(buf)); // Pad with 0xFF

buf[0] = 0x00; // Report ID

memcpy(&buf[1], data, data_length);

pUsb->outTransfer(bAddress, epInfo[epInterruptOutIndex].epAddr, sizeof(buf), buf);
```

If that does not work, then try to send the data directly on the control endpoint:

```C++
// bmRequest = Host to device (0x00) | Class (0x20) | Interface (0x01) = 0x21, bRequest = Set Report (0x09), Report ID (0x00), Report Type (Output 0x02), interface (0x00), datalength, datalength, data)
pUsb->ctrlReq(bAddress, epInfo[0].epAddr, bmREQ_HID_OUT, HID_REQUEST_SET_REPORT, 0x00, 0x02, 0x00, data_length, data_length, data, NULL);
```

dennisfrett, 2021-02-04:
I played around with it a bit further, and figured it out by comparing with Wireshark.
The MiniDSP only accepts 64-byte messages, anything else, and you don't get a response.

Thanks for the info, you helped a lot! 

Got a big part of the driver working already, will create a PR when it's done.

closed 2021-02-04 by dennisfrett