felis/USB_Host_Shield_2.0 #597: What is the purpose of this second for-loop?

liudr, 2021-01-29:
I'm trying to understand the code. I know that the first for-loop processes any port that has changes. But what does the second for-loop do? If the port is not disabled, emulate a change of port connection status? Why? Thanks.

https://github.com/felis/USB_Host_Shield_2.0/blob/b4bbfecd093b0032fa8795369ed914854a8b706e/usbhub.cpp#L283

Lauszus, 2021-01-31:
@xxxajk since you where the one that added the second for-loop: <https://github.com/felis/USB_Host_Shield_2.0/commit/2be575ffc0601dbc82c23d339865810d32e5841b?w=1#diff-2087ad00d2096e73d2f4167b8e5386ab82ff9a8afaafcb05062043420153f9d3R226>, then maybe you can explain it?

xxxajk, 2021-01-31:
Allows a connection event to be detected instead of a reset event, which looks similar.

liudr, 2021-01-31:
OK thanks. So the first for-loop iterates over all ports that have status changes, get the port's status by calling GetPortStatus(). It then calls PortStatusChange() with the evt to process the change. In the processing, a port with C_PORT_CONNECTION gets reset. If a port's been reset, CheckHubStatus() returns immediately with zero without checking the rest of the ports.

Question: Is this logic trying to reset one port at a time for enumeration and prevent two or more ports from getting reset simultaneously? If two devices are both connected (internal devices or initial booting), then calling PortStatusChange() on the second port will cause an immediate return of 0 because bResetInitiated has been set. This makes the first for-loop continue and complete despite the other ports have changes.

So the next for-loop iterates over all valid ports again regardless there's any changes. It gets port status and decide if there's anything. Since the PortStatusChange() already hand shook the C_PORT_CONNECTION, it has to emulate a C_PORT_CONNECTION for every port. I don't understand why this has to happen. I know the bmHUB_PORT_EVENT_CONNECT includes either C_PORT_CONNECTION, or POWER, or CONNECTION.

...Gap of knowledge due to not understanding why the emulated C_PORT...

Once port reset completes, calling PortStatusChange() will hand shake the reset complete, configures the new device with address (probably), resets bResetInitiated, and returns zero. Then any more port with C_PORT will be reset, one at a time.

I hope I got the gist of things.

BTW, why the "port<8" in the first for-loop but "port<=bNbrPorts" in the second? If the hub only has bNbrPorts ports, why checking all the way up to 7?

xxxajk, 2021-01-31:
Answer 1: Yes, you only enumerate one at a time. The logical flow is as follows:

1. Connection detected *somewhere* Could be root, could be a hub.
2. Set up and get very basic information, and send a reset, because someone at microsoft screwed up in 1998, and many devices expect it to reset a second time.
3. Set up the address, AGAIN, because of the reset and above bug in the spec.
4. Enumerate, etc.

The problem that the code solves is that a reset and a plug event both look exactly the same.

Answer 2: There are never > 8 ports per hub. They are numbered from 0 to 7.

liudr, 2021-02-01:
Thanks for the response. Is there any reference to the "1998" screw-up I can read about? I have only worked on limited hubs and devices so I don't know anything that needs to be reset twice. I also didn't have any hubs that show wrong number of ports. Do you mean a hub with say 7 ports only say it has 4 ports? Shouldn't that be considered defected? I've seen 7-port hubs with a 4-port hub chained with another 4 port hub. Probably not what you mean. Any information is highly appreciated.