felis/USB_Host_Shield_2.0 #600: USB Host Shield based on MAX3421E interfacing with ATTiny 1634 problems

akashchandrashrivastava, 2021-02-06:
I am trying to interface the USB host shield with ATTiny1634MUR in QFN20 package and I am facing problems. I am using (https://github.com/SpenceKonde/ATTinyCore ) repo for ATTiny1634 and (https://github.com/felis/USB_Host_Shield_2.0) library for the USB host shield and have added the pin and board definition in USB libraries.

My USBcore.h file looks like this-

/* shield pins. First parameter - SS pin, second parameter - INT pin */
#ifdef BOARD_BLACK_WIDDOW
typedef MAX3421e<P6, P3> MAX3421E; // Black Widow
#elif defined(CORE_TEENSY) && (defined(AVR_AT90USB646) || defined(AVR_AT90USB1286))
#if EXT_RAM
typedef MAX3421e<P20, P7> MAX3421E; // Teensy++ 2.0 with XMEM2
#else
typedef MAX3421e<P9, P8> MAX3421E; // Teensy++ 1.0 and 2.0
#endif
#elif defined(BOARD_MEGA_ADK)
typedef MAX3421e<P53, P54> MAX3421E; // Arduino Mega ADK
#elif defined(ARDUINO_AVR_BALANDUINO)
typedef MAX3421e<P20, P19> MAX3421E; // Balanduino
#elif defined(ARDUINO_X86) && PLATFORM_ID == 0x06
typedef MAX3421e<P3, P2> MAX3421E; // The Intel Galileo supports much faster read and write speed at pin 2 and 3
#elif defined(ESP8266)
typedef MAX3421e<P15, P5> MAX3421E; // ESP8266 boards
#elif defined(ESP32)
typedef MAX3421e<P5, P17> MAX3421E; // ESP32 boards
#elif (defined(AVR_ATmega644P) || defined(AVR_ATmega1284P))
typedef MAX3421e<Pb4, Pb3> MAX3421E;
#elif (defined(AVR_ATtiny1634))
typedef MAX3421e<Pa6,Pa5> MAX3421E;
#else
typedef MAX3421e<P10, P9> MAX3421E; // Official Arduinos (UNO, Duemilanove, Mega, 2560, Leonardo, Due etc.), Intel Edison, Intel Galileo 2 or Teensy 2.0 and 3.x
#endif

My avrpins.h looks like this-

#elif defined(AVR_ATtiny1634)
#define P1 Pa6
#define P2 Pa5
#define P3 Pa4
#define P4 Pa3
#define P5 Pa2
#define P6 Pa1
#define P7 Pa0
//#define P8 Pa4
//#define P9 Pa4
#define P10 Pc5
#define P11 Pc4
#define P12 Pc3
#define P13 Pc2
#define P14 Pc1
#define P15 Pc0
#define P16 Pb3
#define P17 Pb2
#define P18 Pb1
#define P19 Pb0
#define P20 Pa7

My USBhost.h looks like this-

/* SPI pin definitions. see avrpins.h */
#if defined(PIN_SPI_SCK) && defined(PIN_SPI_MOSI) && defined(PIN_SPI_MISO) && defined(PIN_SPI_SS)
// Use pin defines: arduino/Arduino#4814
// Based on: https://www.mikeash.com/pyblog/friday-qa-2015-03-20-preprocessor-abuse-and-optional-parentheses.html
#define NOTHING_EXTRACT
#define EXTRACT(...) EXTRACT VA_ARGS
#define PASTE(x, ...) x ## VA_ARGS
#define EVALUATING_PASTE(x, ...) PASTE(x, VA_ARGS)
#define UNPAREN(x) EVALUATING_PASTE(NOTHING_, EXTRACT x)
#define APPEND_PIN(pin) P ## pin // Appends the pin to 'P', e.g. 1 becomes P1
#define MAKE_PIN(x) EVALUATING_PASTE(APPEND_, PIN(UNPAREN(x)))
typedef SPi< MAKE_PIN(PIN_SPI_SCK), MAKE_PIN(PIN_SPI_MOSI), MAKE_PIN(PIN_SPI_MISO), MAKE_PIN(PIN_SPI_SS) > spi;
#undef MAKE_PIN
[... 191 lines cut ...]
#elif defined(AVR_ATtiny1634)
typedef SPi<Pc1,Pb1,Pb2,Pa6>spi;
#else
#error "No SPI entry in usbhost.h"
#endif

Is there anything else that has to be added to these libraries/header/C files. If anyone could suggest how to go forward or give some advice, it would be really helpful.

Thanks,
Akash.
typedef SPi< P16, P18, P17, P10 > spi;
#elif defined(ESP8266)
typedef SPi< P14, P13, P12, P15 > spi;
#elif defined(ESP32)
typedef SPi< P18, P23, P19, P5 > spi;
#elif defined(AVR_ATtiny1634)
typedef SPi<Pc1,Pb1,Pb2,Pa6>spi;
#else
#error "No SPI entry in usbhost.h"
#endif

Is there anything else that has to be added to these libraries/header/C files. If anyone could suggest how to go forward or give some advice, it would be really helpful.

Thanks,
Akash.

Lauszus, 2021-02-06:
Looking at https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_1634.md, then you entry in avrpins.h is wrong. It should be:

```C++
#define P0 Pb0
#define P1 Pa7
#define P2 Pa6
#define P3 Pa5
#define P4 Pa4
#define P5 Pa3
#define P6 Pa2
#define P7 Pa1
#define P8 Pa0
#define P9 Pc5
#define P10 Pc4
#define P11 Pc2
#define P12 Pc1
#define P13 Pc0
#define P14 Pb3
#define P15 Pb2
#define P16 Pb1
#define P17 Pc3
```

Your entry in USBHost.h should be:

```C++
typedef SPi< Pc1, Pb1, Pb2, Pb3 > spi;
```
Where `Pb3` i.e. the SS pin can be any pin you like, as it does not have a native hardware SS pin.

Regarding the entry in USBCore.h, then it can be any pins you like, but please use the pin defines i.e. `P0`, `P1` etc. Please see: https://github.com/felis/USB_Host_Shield_2.0/#interface-modifications.

FYI format you comments, as it makes them much easier to read and understand. Please see: https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks. I have formatted your comment for you.

akashchandrashrivastava, 2021-02-08:
I have made the changes as you suggested. On an oscilloscope I am getting all the pulses like MOSI,MISO (DI,DO) ,SCK, SS.
On the the ATTiny, I am printing the USB data using software serial. However, when I try to connect USB device to the Host shield, the ATTitny controller most likely gets reset as it stops sending serial data. 
When I remove the connection which ties the RESETS of both (USBHost-MAX3421E and ATTiny), and then reset the ATTiny,  it starts sending serial data again. And when I connect back the RESETs of both (USBHost-MAX3421E and ATTiny), the ATTiny continues to print data on serial but USB device does not get connected.

xxxajk, 2021-02-11:
3 possible causes I can think of, in order of how common I have seen them be the cause.

1. Not enough power, which triggers a brown-out condition. Use higher amperage supply, or if using a USB port to power the ATTiny, use a powered hub. Watch on an oscilloscope for power sags. This is usually the problem.
2. Out of memory. (kind of self explanatory)
3. This is related to the first issue. Use a soft-start USB power switch, so that it doesn't cause a huge draw, and also powercycles the device. This has helped with some devices I have, since it won't power the device until power has settled. Additionally, you can power off the device in software, and save power for battery powered devices.

akashchandrashrivastava, 2021-02-11:
It was a memory issue, removed software serial. The controller was halting because it was out of RAM.

closed 2021-02-11 by akashchandrashrivastava

Lauszus, 2021-02-24:
Please consider sending PR with your changes, so other people can benefit from your work :)