felis/USB_Host_Shield_2.0 #601: Keyboard is not working

Somull, 2021-02-09:
Hello, I've seen this posted already around but none of the solutions proposed has solved my issue.

I'm on Arduino Uno and I've no inputs from my keyboard (Logitech G413). I thought it was a power issue due to backlighted keyboard (5V 500mA) but even an AC adapter hasn't solved the problem (the whole setup drains only around 250mA from AC though). Unfortunatly I got 2 keyboard of the same model so I can't try another one.

My starting point was USBHIDBootKbd from Tygo-bear's Multiboard as well as the one provided in library examples, none of theese printed anything in serial window after the "Start" message.

tmk, 2021-02-09:
I had STALL error when reading descriptor with some keyboards including Logitech G213.

Try these patches. You may need one of them or both if you have same STALL error.
https://github.com/tmk/USB_Host_Shield_2.0/commit/2e4b49c8547b7d2a237dcefa1cc00e4a6c00d28f
https://github.com/tmk/USB_Host_Shield_2.0/commit/15a323c3644ad0e54f89e27a1ef25cb4054db195

Somull, 2021-02-09:
that's great thanks, it solved my issue.

closed 2021-02-09 by Somull