felis/USB_Host_Shield_2.0 #607: How to recover from OUT-NAK?

YuuichiAkagawa, 2021-03-06:
SNDFIFO is not cleared when USB::OutTransfer() exits when it reaches nak_limit.
Probably, when the next data is sent in this state, SNDFIFO becomes a double buffer operation.
https://youtu.be/ZxuxnGrov9U

[image]

When HRSLT returns non-zero, writing 0 to SNDBC seems to clear the SNDFIFO.
https://github.com/felis/USB_Host_Shield_2.0/compare/master...YuuichiAkagawa:pr_outnak_fix
https://github.com/YuuichiAkagawa/USB_Host_Shield_2.0/blob/c56eee9f163b25c2e4be76bd70f44bbf3c17cb8f/Usb.cpp#L390-L396

https://youtu.be/ZxuxnGrov9U?t=29
[image]

Lauszus, 2021-03-25:
Good find. Can you send a PR with the fix?

YuuichiAkagawa, 2021-03-26:
Yes I will. 
However, I have only tested MIDI driver + bulk transfer. Can anyone test in other cases?

Lauszus, 2021-05-02:
Fixed in #617

closed 2021-05-02 by Lauszus