felis/USB_Host_Shield_2.0 #609: Nintendo Switch Pro Controller (with rumble/haptics) ?

zezba9000, 2021-03-12:
I found a fork of this project that added some Switch Pro Controller support: https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTXBOX/blob/master/src/SWProBTParser.cpp

However doesn't look like it has haptics or USB support (only BT).

Is there any other work done for this device? I can't seem to find to much.

Lauszus, 2021-03-25:
I haven't worked on it, as I do not have such a controller. So I would recommend either doing it yourself or sent me a controller and I'll add support for it.

zezba9000, 2021-03-26:
I would be happy to buy you a controller to add support for.
**I'll send you an email for this.**

Also what strategy are you using to capture data being sent from a console to the controller (such as haptic data)?
I wouldn't mind adding some support for missing stuff down the road. Capturing what the controller sends is easy but what a console sends I'm not sure what hardware is best for this?

**One more question if you know.** So I started to port some PS3 driver code to run native on Windows but the PS3 controller won't receive or send any data. I can't send it the PS3 init command like the Arduino does. Not sure what I'm missing for that.

Lauszus, 2021-03-26:
> I'll send you an email for this.

I just replied to your email :)

> Also what strategy are you using to capture data being sent from a console to the controller (such as haptic data)?
> I wouldn't mind adding some support for missing stuff down the road. Capturing what the controller sends is easy but what a console sends I'm not sure what hardware is best for this?

For decoding the data I usually just decode the buttons etc manually and then write the code. For setting LEDs etc then I am usually able to find some an open source driver for it that has already implemented it (usually a Linux driver). Fx in the case of the PS5 diver Sony has actually made an official driver for Linux: https://patchwork.kernel.org/project/linux-input/cover/20201219062336.72568-1-roderick@gaikai.com/.

I have also used Wireshark for looking at how my computer handles the USB and Bluetooth communication.

Finally other people have used the Beagle USB 12: https://www.totalphase.com/products/beagle-usb12/. This is needed if you want to sniff what a console sends to a controller. However I do not own one myself.

> One more question if you know. So I started to port some PS3 driver code to run native on Windows but the PS3 controller won't receive or send any data. I can't send it the PS3 init command like the Arduino does. Not sure what I'm missing for that.

Unfortunately you do have to send that special command either via USB: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.cpp#L488-L497 or Bluetooth: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3BT.cpp#L583-L593 or the controller will not send out any data. I guess Sony originally did this to "protect" it or whatever, but luckily they learned their lesson with the PS4 and PS5 controller which acts as a regular joystick. I do not see the reason for "protecting" it in that way, as people will always be able to reverse engineer it. I guess they also saw how Microsoft was able to sell Xbox controllers to PC gamers and wanted to get into that market as well.

zezba9000, 2021-03-27:
**Here is a HID linux driver that supports rumble, BT & USB:** https://github.com/Dan611/hid-procon

----------------
The PS3 stuff on Windows HID is odd. I may test on Linux to see if its just a Windows specific issue blocking me from sending the controller the special init command.

Also another odd thing with PS3 is I have a Special GameStop controller that is for PS3 but maps totally different from normal PS3 controllers and doesn't require special init command. I will confirm it does in fact work on PS3 and may add support for this special PS3 controller type. It has a VID-0x0E6F & PID-0x0268

zezba9000, 2021-05-06:
All features tested. Closing this issue.

PR #632 (2021-05-09): Added support for the Switch Pro controller
Fixes #609

* [x] Bluetooth driver
* [x] Control LEDs and rumble
* [x] USB driver
* [x] Use full report, so IMU data is included
closed 2021-05-06 by zezba9000