felis/USB_Host_Shield_2.0 #610: Original Xbox controller won't poll (hrJERR)

dmadison, 2021-03-13:
I'm trying to use one of the original Xbox "Duke" controllers (VID 0x045E, PID 0x0202) with the library and it's not working. Using [the XBOXOLD example](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Xbox/XBOXOLD/XBOXOLD.ino) the controller appears to initialize but never receives update data from the control surfaces. I modified the `inTransfer` call in the `Poll` function to grab the return code and it looks like it's failing with `hrJERR` (0x0D). If I'm reading the datasheet correctly that means the device responded with a J state instead of the requested data packet, so I'm guessing the host may not be properly configured for the device?

With `ENABLE_UHS_DEBUGGING 1` the full output from the example is:

```
XBOX Library Started
Xbox Controller Connected
```

The controller works fine when connected to a PC [with a third party driver](https://github.com/emoose/Xb2XInput). I'm testing with a Teensy 3.2 connected to a mini host shield, modified to work with 5V on the bus. The hardware setup works with a number of other devices in the library so I'm doubtful it's to blame.

Do you have any recommendations for things I might try? I can provide a logic analyzer output of the D+/D- lines if that would be beneficial.

sko4in, 2021-03-25:
I second that. OG Duke and Japanese S controller both don't work, they pass Init and then Buffer from Poll is just empty.

08:35:23.628 -> XBOX Library Started
08:35:30.766 -> VID:1118
08:35:30.766 -> PID:514
08:35:30.766 -> RCode:0
08:35:30.766 -> Poll Enable:1
08:35:30.766 -> Poll Count:1
08:35:30.766 -> Address:9
08:35:30.766 -> Ep Address:1
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
08:35:30.766 -> 0
^^^^- all the zeroes are from readBuf;
Everything works (after some tweaking for ReadHatValues) on OG Small Controller.

Lauszus, 2021-03-25:
@dmadison start by enabled `EXTRADEBUG` and `PRINTREPORT`: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXOLD.cpp#L20-L21. This give you some more information.

Try to experiment with the `BUFFER_SIZE` as well: https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXOLD.cpp#L248

Also you could try to change it to "Windows style enumeration" - see: https://github.com/felis/USB_Host_Shield_2.0/pull/566/files#diff-b4ada7664f8bd69e49cd021e1715a5e8a88da28a2d043cc4bad800358d62d57cL150-R195

dmadison, 2021-03-27:
Thanks for the advice @Lauszus.  Here is the output with `ENABLE_UHS_DEBUGGING`, `EXTRADEBUG` and `PRINTREPORT` enabled:

```
XBOX Library Started
XBOXUSB Init
Addr: 09
Xbox Controller Connected
```
Each following packet is filled entirely with 0s, presumably because no data is ever returned.

Increasing the `BUFFER_SIZE` appears to have no effect.

---

With the "Windows style enumeration", I copied a block of lines from that commit and stuck them before the first device descriptor request ([line 109](https://github.com/felis/USB_Host_Shield_2.0/blob/a321ad36ab4dd9ea000ffa809532a937ffca06d0/XBOXOLD.cpp#L109)):

```
        // First Device Descriptor Request (Initially first 8 bytes)
        // https://techcommunity.microsoft.com/t5/microsoft-usb-blog/how-does-usb-stack-enumerate-a-device/ba-p/270685#_First_Device_Descriptor
        rcode = pUsb->getDevDescr( 0, 0, 8, (uint8_t*)buf );

        // Restore p->epinfo
        p->epinfo = oldep_ptr;

        if( rcode ){
                goto FailGetDevDescr;
        }

        // Allocate new address according to device class
        bAddress = addrPool.AllocAddress(parent, false, port);
        if (!bAddress) {
                return USB_ERROR_OUT_OF_ADDRESS_SPACE_IN_POOL;
        }

        // Extract Max Packet Size from device descriptor
        epInfo[0].maxPktSize = udd->bMaxPacketSize0;

        // Assign new address to the device
        rcode = pUsb->setAddr( 0, 0, bAddress );
        if (rcode) {
                p->lowspeed = false;
                addrPool.FreeAddress(bAddress);
                bAddress = 0;
                return rcode;
        }//if (rcode...
        USBTRACE2("Addr:", bAddress);

        p->lowspeed = false;

        //get pointer to assigned address record
        p = addrPool.GetUsbDevicePtr(bAddress);
        if (!p) {
                return USB_ERROR_ADDRESS_NOT_FOUND_IN_POOL;
        }
        p->lowspeed = lowspeed;
```

The second attempt to get the device descriptor fails with an rcode of `0x0D`, which I believe is again that `hrJERR` state.

It was working with a custom driver on my PC. I'll see if I can find some time to hook up a logic analyzer and compare the two initialization sequences to see where it's going wrong.