felis/USB_Host_Shield_2.0 #612: USB HOST Reset

GoldzGit, 2021-03-15:
The device must control a device with a USB interface.
The device is assembled on ESP8266 and USB HOST mini (DuinoFUN USBmini V2.0 Aug.16,2014).
I am using your library (acm_terminal). Everything is fine, but in order for the device to work with my device, it must be rebooted.
I cannot do this. The device is assembled on ADUINO MEGA. I cannot change its firmware.
I think that if you do RESET USB HOST, then the device will reboot.
It didn't work for me. When controlling any ARDUINO via cable, I set COM PORT to DTR = true then DTR = false and Arduino reboots.
Perhaps this can be done using your library at the USB HOST level.

In connection with this problem, I am in great despondency, really three months of my life have passed in vain, and I am already 60 years old. Every day is dear.
I ask you to help me solve my problem.

P.S English translation made by Google

GoldzGit, 2021-03-21:
uint8_t resetHost()
{
  uint8_t rcode;
  rcode = Acm.SetControlLineState(1);
  rcode = Acm.SetControlLineState(0);
  rcode = Acm.SetControlLineState(1);
  if (rcode)
  {
    ErrorMessage<uint8_t>(PSTR("SetControlLineState"), rcode);
    return rcode;
  }

xxxajk, 2021-03-22:
Any ESP I have encountered uses both DTR and RTS together to reboot it.
This is because one of the lines (depending on the implementation) uses one of them or both to enter bootloader mode.

On AVR avrdude uses both as well, since some older designs used RTS instead of DTR, some need both at the same time at the same level... it's a mix of legacy designs, and some want the level actually reversed to make matters worse. On top of all of that, some designs actually check for a change in baud rates.

Your best bet, is to use an oscilloscope to verify you are getting things correct.

GoldzGit, 2021-03-26:
Hello
This works for sure with the cdcacm.h library
void resetHost ()
{
  
     uint8_t rcode;
     rcode = Acm.SetControlLineState (3);
     delay (100);
     rcode = Acm.SetControlLineState (0);
     delay (100);
     rcode = Acm.SetControlLineState (3);
     delay (100);
     if (rcode)
     {
     ErrorMessage <uint8_t> (PSTR ("SetControlLineState"), rcode);

     }
}
I can not reset with the cdcftdi.h library
I connect a 3d printer and I need to reset the printer
I ask for help in solving this problem.