felis/USB_Host_Shield_2.0 #616: Cannot send midi messages to a Nux MG300

wo2507, 2021-03-24:
hello,
i'm trying to send midi messages to a Nux mg-300 (guitar effect).
the code is work well with Zoom MS50G, but not work to my Nux MG300

here is my test code 

`
 

#include <Usb.h>
#include <usbhub.h>
#include <usbh_midi.h>

USB Usb;
USBH_MIDI Midi(&Usb);
// this is my assumptions
uint8_t Patch1A = 0x00; // 0x00 is pactch 1A on nux mg300
uint8_t Patch2A = 0x01; // 0x01 is patch 2A on nux mg300

void setup()
{ 
  pinMode( 7, OUTPUT);  // For the sheild
  digitalWrite( 7, HIGH); // for the sheild
  if (Usb.Init() == -1)
  {   
    while(1);               // Halt
  }
  delay(200);  
}

void loop()
{
   //test to nux mg300
   SendMIDIPC(Patch1A);
   delay(1000);   
   SendMIDIPC(Patch2A);
   delay(1000); 
   
   //test to zoom ms50g
   /*tunerOn();
   delay(1000);
   tunerOff();
   delay(1000);*/
       
}

// Patch Changer

void SendMIDIPC(uint8_t number)
{
  Usb.Task();  
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {       
    uint8_t buf[3];                 // Construct the midi message (3 bytes)
    buf[0]=0xB0;                    // Control change 
    buf[1]=0x3C;                    // 0x3C is #CC60 for change the patch on nux mg300 
    buf[2]=number;                  // number is patch number (write in hex above)
    Midi.SendData(buf);             // Send the message    
    delay(10);
  }   
}

///////////=========== this is for activated/deactivated tuner on zoom ms50g ===========////////////

/*void tunerOn()
{
  Usb.Task();  
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {       
    uint8_t buf[3];                 // Construct the midi message (3 bytes)
    buf[0]=0xB0;                 // control Change 
    buf[1]=0x4A;               // CC 74
    buf[2]=0x40;            // tuner On , 40 = CC Value 64
    Midi.SendData(buf);         
    delay(10);
  }   
}

void tunerOff()
{
  Usb.Task();  
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {       
    uint8_t buf[3];                 // Construct the midi message (3 bytes)
    buf[0]=0xB0;                 //  control Change 
    buf[1]=0x4A;               // CC 74
    buf[2]=0x00;            // tuner off
    Midi.SendData(buf);          
    delay(10);
  }   
}
*/

`

YuuichiAkagawa, 2021-03-24:
Run https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/USB_MIDI_desc/USB_MIDI_desc.ino

YuuichiAkagawa, 2021-03-24:
reference https://www.vguitarforums.com/smf/index.php?PHPSESSID=bbda9e64154c4340ebc6ce2011ed2a9d&topic=29139.msg212388#msg212388

eggyerdiyus271013, 2021-04-03:
Hiii, this issue solved already ?

PR #634 (2021-05-09): Update MIDI driver v0.6.0
2021.5.9 (0.6.0)

- Change configuration descriptor parser. Supports large descriptors. (Fix #616)
- Fixed an issue when the endpoint size exceeded 64 bytes. (Fix #616)
- Add OnInit() callback
- Add a predefined macro "USBH_MIDI_VERSION"
- Changed the method of deriving the message length
- Refine the examples
closed 2021-04-04 by wo2507