felis/USB_Host_Shield_2.0 #619: How to send custom non-HID USB m using this library?

bsperryn, 2021-03-28:
I'm looking at using my Arduino as a proxy between a PS4 and a PS4 controller.

To do this I need to forward the authentication process between the 2 devices as described here:
https://forum.gimx.fr/viewtopic.php?t=2384&start=10

- Console sends a nonce in 5 packets type 0xF0
- DS4 sends one ACK packet type 0xF2 that's its prepping the data (probably because RSA is compute-heady and that microcontroller on the gamepad needs some time)
- DS4 sends one ACK packet type 0xF2, but this time, removes the flag, indicating that its ready to send commands.
- DS4 sends 19 packets type 0xF1.

Challenge:

```
struct ps4_challenge {
 	unsigned char nonce[0x100];
};
```
Response:
```
struct ds4_response {
	unsigned char signature[0x100];
	unsigned char serial_num[0x10];
	unsigned char n[0x100];
	unsigned char e[0x100];
	unsigned char casig[0x100];
};
```
signature is a PSS signature of the nonce, signed by the private key of DS4
serial_num is the controller/cert serial number.
n is the Public Key's prime number
e is the Public Key's exponent
casig is a PSS signature (signed by Sony's CA private key) of the serial_num, n and e. This is what prevents you from generating your own keys and this is why you need a valid DS4 dump.

 
I will use another library and USB port to help receive the challenge from the PS4 and forward the responses from the controller back.

I was hoping you could help me with how best to forward this non HID challenge from the PS4 on to the controller and then how to receive the 3 non-HID response's from the controller?

gdsports, 2021-03-29:
USB MITM between PS4 and DS4 has been done apparently. I have never tried it.

https://dsremap.readthedocs.io/en/latest/index.html

bsperryn, 2021-03-29:
Thank you for this, looks promising!

gdsports, 2021-03-29:
I do not know of a project similar to dsremap for Xbox.

The GIMX project supports Xbox and PS4 but it runs on a PC or Raspberry Pi. It uses a Pro Micro to talk to the console.

https://gimx.fr/wiki/index.php?title=Main_Page

Ryzee119, 2021-03-30:
https://github.com/Ryzee119/ogx360
I made this a few years ago. For original xbox (not xbox 360) but uses this lib for host control and LUFA for device emulation

bsperryn, 2021-03-30:
Hmmm, unfortunately LUFA is only for older AVR's... I'd really like to get this working on a newer chip like the Samd21. (I'm using a IOT Nano at the moment) Any idea on how I might go about that?

Ryzee119, 2021-03-30:
Ive used TinyUSB for xbox controller emulation. It supports samd21.

bsperryn, 2021-04-02:
@Ryzee119 Which board did you use TinyUsb on? I'm having a problem getting it working with my nano IOT...

Ryzee119, 2021-04-02:
It happened to be a Teensy4 
 https://github.com/Ryzee119/ogx360_t4