felis/USB_Host_Shield_2.0 #625: Undef Keypad in USBHID_desc

hi-im-haz3l, 2021-04-17:
Hello, I've seen this posted already around but none of the solutions proposed has solved my issue.

I'm on Arduino Pro Mini and I've no inputs from my mechanical keyboard (Akko 3087S). I thought it was a protocol issue and the driver on windows gonna translate it back but when I capture its packets using wireshark, everything is normal (the "a" key return 00 00 04 00 00 00 00 00). Fortunatly I got 2 keyboards and the other one work fine. I've tried to fine the different between them but I've no luck so far.

My starting point was USBHIDBootKbd, It printed nothing in serial window after the "Start" message. After that comes USBHID_desc, that is the only sketch in the whole HID folder that gave anything beside "Start". But compare to my other keyboard, every keystrokes printed the same response but there is "Undef Keypad" instead of "Keypad".

I'm not using an USB Hub.

 **_tl;dr_**
**All keystrokes not recognized.
USBHID_desc output:** https://pastebin.com/B2LPeL5f
**USB_desc output:** https://pastebin.com/vt5J0V1W

Should I try to edit the code someway or it's probable that this keyboard never gonna work?
Thanks.

Lee

Z4ck2021, 2021-07-16:
is there any solution for this issue , I'm also the same  undef keypad  but the data return it's correct.
some how cannot be  interpret as  asci code during  HidKbd  test.

Z4ck2021, 2021-07-16:
Start
0000: 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 
0010: 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 
0020: 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06 
0030: 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 
0040: 05 0C 09 00 15 80 25 7F 95 40 75 08 B1 02 C0 
Usage Page Gen Desktop Ctrls(01)
Usage Keypad
Collection Application
Usage Page Kbrd/Keypad(07)
Usage Min(E0)
Usage Max(E7)
Logical Min(00)
Logical Max(01)
Report Size(01)
Report Count(08)
Input(00000010)
Report Count(01)
Report Size(08)
Input(00000001)
Report Count(05)
Report Size(01)
Usage Page LEDs(08)
Usage Min(01)
Usage Max(05)
Output(00000010)
Report Count(01)
Report Size(03)
Output(00000001)
Report Count(06)
Report Size(08)
Logical Min(00)
Logical Max(FF00)
Usage Page Kbrd/Keypad(07)
Usage Min(00)
Usage Max(FF00)
Input(00000000)
Usage Page Consumer(0C)
Usage Undef
Logical Min(80)
Logical Max(7F)
Report Count(40)
Report Size(08)
Feature(00000010)
End Collection⸮Keypad(00)(00)(00)(00)(00)(00)(00)(00)
(00)
(04)(00)(00)(00)(00)(00)
 Undef Keypad(00)(00)(00)(00)(00)(00)(00)(00)
(00)
(00)(00)(00)(00)(00)(00)
 Undef

tmk, 2021-07-16:
As for USBHID_desc,

GetReportDescr() requests always 128 bytes for Report descripotor, this may cause problem with some devices like this keyboard. The function should request with real length of the descriptor intead of the fixed 128. 

https://github.com/felis/USB_Host_Shield_2.0/blob/969eabb8d7979c4cf76a02677f433d2ef5af7735/usbhid.cpp#L39

So USBHID_desc read extra bytes of unknown invalid data  and failed to parse at the end part. But anyway, it displays Report descriptor of your keyboard except for the garbage.

And there is another problem, you can't get other Report descriptors than one of interface0 with USBHID_desc. Notice that the keyobard has Report descriptor on interface1 as well. 

https://github.com/felis/USB_Host_Shield_2.0/blob/13950453c9c59acc92fab71cd8603d2ec4df8870/examples/HID/USBHID_desc/USBHID_desc.ino#L32

If you want to inspect descriptors for debug you can use tools like lsusb or wireshark.

Z4ck2021, 2021-07-16:
So  by  right if i  change on those two line  inside usbhid.cpp and  sketch for  hiddescriptor it will work ?

BTW thanks for your advice  @tmk .  I'll try  and revert back soon.

Z4ck2021, 2021-07-16:
Still cannot be detect on  my Serial Monitor;   it cannot run  for the  Hidkbd-test.ino  sketch.  ( conversion  hex to ASCII)
I hope someone can figure out how to resolve this issue.