felis/USB_Host_Shield_2.0 #627: Xbox One S controller does not pair

gerahamm, 2021-04-21:
**Environment**

- Arduino Nano
- [Mini Usb Host Shield](https://www.aliexpress.com/item/32842815739.html)
- [BT Dongle 1](https://www.aliexpress.com/item/4000414075718.html)
- [BT Dongle 2](https://www.aliexpress.com/item/32787640185.html)

The Usb Host Shield has been tested with a wired Xbox 360 controller and the available example code. Works perfectly.

While using the example code for the Xbox One S Controller ([link](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Xbox/XBOXONESBT/XBOXONESBT.ino)) with any of the given BT dongles I just receive the following output:

```
Xbox One S Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
Dongle supports secure simple pairing (controller support)
```

Other BT dongle has different address.

Tried the following:

1. Removing the BT dongle and connecting the controller via USB produces a lot of unreadable output (strange symbols)
2. External power supply for more power
3. USB hub

Unfortunately without any success.

MasterJubei, 2021-04-25:
I had a lot of problems with the knockoff mini usb host shield, mine looks just like that. I bought two of them, they experience a lot of intermittent problems. They kept freezing my board. The full sized usb host shields all over amazon work just fine though. 

With that said, even if you do get it working, the xbox one s controller is fairly broken in its current state, I would recommend using a PS4 controller since I know that works just fine.

With the Xbox One S controller you have to re-pair every time unless you use the patch I have in a PR request. I also found a lot of the buttons are not mapped correctly, at least on my revision of the controller. The data in XBOXSBT.xboxOneSData.btn is mapped incorrectly. I am trying to fix that, although I have a feeling the data might be different depending on controller locale/revision since mine is very different than the stock code.

HisashiKato, 2021-04-25:
Packets sent via Bluetooth from the Xbox One S controller (MODEL 1708) have changed as follows with the firmware update (4.8.1923.0). 
https://github.com/atar-axis/xpadneo/blob/master/docs/descriptors/xb1s_linux.md

HisashiKato, 2021-04-25:
I made a structure like this for my library. 
https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTXBOX/blob/master/src/XB1SBTParser.h#L89-L126

MasterJubei, 2021-04-25:
> I made a structure like this for my library.
> https://github.com/HisashiKato/USB_Host_Shield_Library_2.0_BTXBOX/blob/master/src/XB1SBTParser.h#L89-L126

Ah great, thanks. My blind findings were looking very strange, although it seemed to still work. Yours is obviously the one to go with though.
```union XboxOneSButtons {
        struct {
                uint8_t dpad : 4;
                uint8_t reserved : 4;

                uint8_t a : 1;
                uint8_t b : 1;
                uint8_t U1 : 1;
                uint8_t x : 1;

                uint8_t y : 1;
                uint8_t U2 : 1;
                uint8_t LB : 1;
                uint8_t RB : 1;
                //uint8_t menu : 1;

                uint8_t U3 : 1;
                uint8_t U4 : 1;
                uint8_t U5 : 1;
                uint8_t menu : 1; //start button
                uint8_t XBOX : 1;
                uint8_t LSB : 1; //LSB
                uint8_t RSB : 1; //RSB
                uint8_t U6 : 1;
                uint8_t SHARE : 1;
                //uint8_t reserved2 : 6;
        } __attribute__((packed));
        uint32_t val : 25;
} __attribute__((packed));```

HisashiKato, 2021-04-25:
With the new firmware, the button data packet has increased by 1 byte, and the contents of the entire button data have changed.

MasterJubei, 2021-04-25:
@gerahamm 
I have not used an Arduino nano, so I have not looked at the details of how that is wired.
But for my setup I had to cut the 5v line on the usb host mini shield. It is default bridged to VCC which is 3.3V.
Cut on the indicated green line.

[image]

gerahamm, 2021-05-05:
@MasterJubei There cannot be a reconnection if the initial pairing does not work at all :)