felis/USB_Host_Shield_2.0 #628: GPIO pins don't work until after first successful USB device connection

mlord, 2021-04-22:
The MAX chip has 8 GPIO-in and 8 GPIO-out pins which can be accessed via this library.  But reading from the pins simply returns all zeros until after the first time the driver enters USB_STATE_RUNNING.  Subsequent state changes are fine -- GPIOs still function then.

mlord, 2021-04-22:
Note:  using the mini USB host shield with ESP32.

mlord, 2021-04-22:
Actually, on further experimentation:  gpioRd() fails (always returns zero) before entering USB_STATE_RUNNING, as reported.   But.. one call to gpioWr() seems to "wake it up" and gpioRd() then works.  This is why the board_qc sketch doesn't notice the problem, because it always does gpioWr() first.

xxxajk, 2021-04-26:
Nice find. I may get some time in a week or two to fix that. Should be an easy fix.