felis/USB_Host_Shield_2.0 #629: audio recording using USB microphone to ESP32.

AisyahJeff95, 2021-04-23:
Is it possible to plug in USB mic into the USB Host shield 2.0, and record stereo audio to the ESP32? 
And then save the .wav file into SD card. I heard that there is no audio driver for this USB module? Or do I have to write my own drivers as mentioned in the USB_HID library?

xxxajk, 2021-04-26:
Possibly, but you would have to write the driver for that.

AisyahJeff95, 2021-04-26:
> Possibly, but you would have to write the driver for that.

hey thanks. So I need to write the driver for USB host shield 2.0 or for the ESP32? If you could share me some links to study these would be great!

I read somewhere that we need to do this:
"After writing the driver, Then, you’d have to re-write the tone() function to communicate with the driver. And, you have to deal with the fact that the ESP32 is normally a USB device and you need it to operate as a USB host controller. Not really practical…"

or should i hack to the USB mic's USB head to make it connect to the analog?

AisyahJeff95, 2021-04-26:
> Possibly, but you would have to write the driver for that.