felis/USB_Host_Shield_2.0 #630: power stops going to controller after period of time.

brian-not-the-messiah, 2021-04-25:
I'm using the mini board with a Nano 33 BLE

Sometimes the controller doesn't get power and connect when first starting up, other times it powers up and connects for a period of time, but then seems to either turn off or stop receiving power for some reason.

Do you guys have any idea why this might be happening or how I might be able to debug this?

xxxajk, 2021-04-26:
Always look on the bright side of life... ;-)

Check your connections, use shortest possible wires. Long wires are a death nail for SPI.
Make sure that you have enough power, use a powered hub on it. USB is very sensitive to power jitter.

bsperryn, 2021-04-27:
;-)

At the moment the connections between the controller and the usb board are 20cm jumpers in a breadboard is that too long/unstable? I'm thinking of soldering much shorter wires directly, does that sound like it should do the trick?

brian-not-the-messiah, 2021-04-27:
;-)

At the moment I've got the controller and the USB board connected using 20cm jumper wires on a breadboard. I'm looking at using much shorter wires directly soldered or connected via a stripboard. Does that sound like it should work?

xxxajk, 2021-04-28:
No, it's not a jitter issue, the problem is that the supply doesn't have the extra capacity.
Don't use a bread board, the capacitance it too high, plus you can end up with cross talk.
20CM is way too long. 5CM is about as far as I would go when using any sort of wire or generic non-sheilded board or cable. The simplest solution is to use short ribbon cable, otherwise you have to use sheilded cable to help with various noise and/or a combination of termination and attenuation to combat reflections

brian-not-the-messiah, 2021-04-29:
Thanks so much for your help.

How about using an external power supply to apply power to the USB port on the USB board? Is that a possibility? With my limited understanding it seems to make sense to me, but since there will be 2 sources of power, I'm not sure how to handle ground. Any advice?

Ultimately, I'm wanting to use this is a commercial product and would prefer to not use external power. The USB ports I'll be connecting to should have enough power as they are USB 3, but the pro micro us only USB 2. 

I've tried taking 5v directly from the SJ1 jumper, but unfortunately this seems to have already had it's current limited by the USB port on the pro micro.

[image]

Would it be easy enough to create a board based on the pro micro that uses USB 3.0 that would allow me to draw enough current from the ports available to me?

xxxajk, 2021-04-30:
Yes, you can do that. All you need to do is have the grounds joined, and a proper fuse, just so, if you were to have a 4 port hub, you would want 2Amps. Honestly, this is much easier if you use a powered hub. However if you want this just as an on-board feature, what I recommend is to desolder F1 to prevent voltage backwash to the new supply (USB is protected, but that pesky diode does drop voltage!!!) and just apply the new 5V supply to RAW and to your host port. :-D

bsperryn, 2021-04-30:
Using your suggestion, what do I do with the grounds in this scenario? Also join them up? If so, where would you join them and what size fuse should I put where? (sorry I'm fairly new to all this)

Also if you wouldn't mind confirming the part circled in red is the one I would need to desolder?

[image]

brian-not-the-messiah, 2021-04-30:
Using your suggestion, what do I do with the grounds in this scenario? Also join them up? If so, where would you join them and what size fuse should I put where? (sorry I'm fairly new to all this)
Also if you wouldn't mind confirming the part circled in red is the one I would need to desolder?

[image]

xxxajk, 2021-05-01:
Yes that's the polyfuse, and yes, just join ground from the new power supply (of course).
two connections, and they are right next to each other, +5v on RAW, and ground to the new 5v power supply.

brian-not-the-messiah, 2021-05-01:
Thank you, I gave that a shot but ran into a problem.

I have now connected the power for the usb to the other side of the
polyfuse.

Will it be possible to get more than 500ma from the 900maUSB 3.0 port of
the computer or will is it limited somehow in the arguing usb2.0 port that
it won’t send more than 500ma even if needed?