felis/USB_Host_Shield_2.0 #631: How do I customize the CDC ACM driver for a specific device (VID 0x2544)?

BastianRuppert, 2021-04-29:
Hello,
I am trying to address USB Devices from Energie Micro AS (Vid 0x2544) based on the CDC ACM driver. These devices provide a serial COM port and hold additional information about company name and device number in the USB Device Settings.

The USB Host Shield 2.0 Mini is connected to an ESP32. 

PrintAllDescriptors and PrintAllAddresses gives me:

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		02
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		2544
Product ID:		0003
Revision ID:		0000
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0043
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			120
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		02
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		05
Type:		24
Contents:	0020010524
Unknown descriptor:
Length:		05
Type:		24
Contents:	0100010424
Unknown descriptor:
Length:		04
Type:		24
Contents:	02020524
Unknown descriptor:
Length:		05
Type:		24
Contents:	0600010705

Endpoint descriptor:
Endpoint address:	82
Attr.:			03
Max.pkt size:		0040
Polling interval:	1515

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		010
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	01
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:0x1(0x0.0x0.0x1)

Adaptations of the driver are necessary to address this kind of devices. The CDC ACM driver based on the file cdcacm.cpp fails because only 3 endpoints are recognized.

I suspect the part from ACM::Init:

`ConfigDescParser< USB_CLASS_COM_AND_CDC_CTRL,
                        CDC_SUBCLASS_ACM,
                        CDC_PROTOCOL_ITU_T_V_250,
                        CP_MASK_COMPARE_CLASS |
                        CP_MASK_COMPARE_SUBCLASS |
                        CP_MASK_COMPARE_PROTOCOL > CdcControlParser(this);`

cannot process the information of the device correctly, bNumEP is 3!

(If I bridge this circumstance
        `if(bNumEP < 3)//4)
                return USB_DEV_CONFIG_ERROR_DEVICE_NOT_SUPPORTED;` 

I can probably address the device correctly but not read it correctly. (Various commands to the device result in the expected number of bytes in the response, but the bytes read out are all zero). )

Can someone give me hints what exactly is going wrong?

How to adjust the driver or the ConfigDescParser <...>CdcControlParser for this type of device?

Thank you very much and best regards,

Bastian.

xxxajk, 2021-05-01:
If your project is small and on an AVR board OR if you are using an Atmega2560 based board OR you are using an ARM based board that isn't using an RTOS, you can try UHS3.0, which does not rely on the number of endpoints for anything.

BastianRuppert, 2021-05-04:
@xxxajk Thank you for your response. 
I will have a look at UHS3.0!
For this issue i can use this ACM class with bNumEP < 3 modification.
Thank you very much.

closed 2021-05-04 by BastianRuppert