felis/USB_Host_Shield_2.0 #633: meaning of Rcode 04

Titibo26, 2021-05-05:
Hi,
I'm using an esp32 to read data from a device equiped with an ftdi chip.

I get informations of the device :
`FTDI Init
18:30:09.637 -> Addr:01
18:30:09.637 -> NC:01
18:30:09.637 -> 0000: 09 02 20 00 01 01 02 20 1E 09 04 00 00 02 FF FF 
18:30:09.637 -> 0010: FF 02 07 05 81 02 40 00 00 07 05 02 02 40 00 00 
18:30:09.637 -> Conf.Val: 01
18:30:09.637 -> Iface Num: 00
18:30:09.684 -> Alt.Set: 00
18:30:09.684 -> Endpoint descriptor:
18:30:09.684 -> Length:		07
18:30:09.684 -> Type:		05
18:30:09.684 -> Address:	81
18:30:09.684 -> Attributes:	02
18:30:09.684 -> MaxPktSize:	0040
18:30:09.684 -> Poll Intrv:	00
18:30:09.684 -> Conf.Val: 01
18:30:09.684 -> Iface Num: 00
18:30:09.684 -> Alt.Set: 00
18:30:09.684 -> Endpoint descriptor:
18:30:09.684 -> Length:		07
18:30:09.684 -> Type:		05
18:30:09.684 -> Address:	02
18:30:09.684 -> Attributes:	02
18:30:09.684 -> MaxPktSize:	0040
18:30:09.684 -> Poll Intrv:	00
18:30:09.684 -> NumEP:03
18:30:09.684 -> Conf:01
18:30:09.684 -> baud_value:C04E
18:30:09.684 -> baud_index:0000
18:30:09.684 -> FTDI configured
`

but after sending data (rcode = 00, its ok), i get 
Ret : 04 
which means there is a problem at reception. 
What is the meaning of this code ? i can't find it anywhere in the lib.
Thanks,

xxxajk, 2021-05-06:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/max3421e.h#L218
0x04 means the device isn't ready yet, so you need to try again.
It's not an error, it's more like saying "I'm busy right now, try later."

Titibo26, 2021-05-06:
Oh thanks, didn't see this code !
I'm using an esp32+usb host mini --> to an usb device equipped with FTDI chip. I have to inspire myself of "USBFTDILoopback" example, am I right ?
 
Does this error (04) can be related to bad communication settings ? the device is supposed to work at [38400, 8bits, no parity, no HS, 1 start bit, 1 stop bit]. I can't see where to set the Start bit. can it be the problem?
`rcode = pftdi->SetBaudRate(38400); 
rcode = pftdi->SetFlowControl(FTDI_SIO_DISABLE_FLOW_CTRL); 
rcode = pftdi->SetData(FTDI_SIO_SET_DATA_PARITY_NONE |FTDI_SIO_SET_DATA_STOP_BITS_1| 8);`

Finnally, Can this error code be caused by wrong command sent ?

Titibo26, 2021-05-07:
Hello,

I'm still investigating my issue and after some debugging, usb allways fall in USB_STATE_ERROR (0x0A).
It goes from 0x11 to 0x20 to 0x40 to 0x50 to 0x51 and then to 0x0A.
Just after 0x51, this condition is true : 
`                                if(rcode != USB_DEV_CONFIG_ERROR_DEVICE_INIT_INCOMPLETE) {
`
Which make it fall in 0x0A.
So the problem seems to be on this line : 
` rcode = Configuring(0, 0, lowspeed);`

Apparently this feature is broken ? do you think problem can comes from here ?

Thanks,