felis/USB_Host_Shield_2.0 #637: USBHIDBootMouse not working with USB MOUSE Logitech G502 HERO.

kamikazaliba01, 2021-05-15:
I'm trying to connect my mouse to the USB HOST SHIELD however it just doesn't show the dX and dY. But using another Logitech mouse, the SuperLight one it works.

nikkoxgonzales, 2021-05-17:
are My3? Even if you can do this route you know you can't easily send signals while a task worker is executing, serial input is not an option.

kamikazaliba01, 2021-05-17:
> are you My3? Even if you can do this route you know you can't easily send signals while a task worker is executing, serial input is not an option.

Yep.
But me cheat it's working again with the host shield.. But there some mouses like the G503 that the host shield just doesn't show output.

nikkoxgonzales, 2021-05-17:
> And my cheat it's working again with the host shield.. But there some

Do you mind showing how you send commands to your leonardo? Still can't,
get it to work. Can I add you on discord?

Regards,

Nikko Gonzales

nikkoxgonzales, 2021-05-21:
Okay, I got the time and I've done it, I've even added my G503. @kamikazaliba01 I don't know if you ignored my message, but you're literally asking for help in Github and you don't want to help others when you have the solution. LOL.

kamikazaliba01, 2021-05-21:
dm in discord. KaMiKaZ#1305

kkh961216, 2021-06-02:
@nikkoxgonzales  @kamikazaliba01  would either of you mind adding me on discord ? I have a question regarding getting my logitech mouse to work with this library

TheGreatLambie#4183

dylanrush, 2021-09-21:
I am trying to get the G502 to work with a USB host shield and I'm not able to get a signal. Is there a workaround?

xxxajk, 2021-09-22:
And you need to cheat at games because you suck :-)

nikkoxgonzales, 2021-09-22:
> And you need to cheat at games because you suck :-)

People cheat because they can. Simple as that. You can say they suck but is it really their fault for devs lack of skills in implementing such checks?

dylanrush, 2021-09-22:
This is funny. I honestly did not consider that this device could be used for cheating but I suppose it could be used for bullet spray macros or whatever. Couldn't you also do that in software?

My actual use case is much more benign. I have four computers in my office all connected 4-way KVM with USB and HDMI: my gaming PC, my work computer and a couple older computers. The oldest is an IBM Aptiva from around 1997. It technically has USB ports but these do not work in DOS or Windows 95. PS/2 is the only protocol that reliably works for this computer. So I am building a bridge from USB to PS/2 for both the keyboard and mouse.

I actually got this to work. Ideally this library would allow me to specify an address when constructing a USB device instance. Instead, when you construct something like an HIDBoot it just connects to the first one. So the order in which these classes are instantiated matters. I also could not get HIDBoot to work with this mouse, but HIDUniversal does.

So the magic incantation was:

```
HIDBoot < USB_HID_PROTOCOL_KEYBOARD | USB_HID_PROTOCOL_MOUSE > HidComposite(&Usb);
HIDUniversal                                    Hid(&Usb);

  HidComposite.SetReportParser(0, &KbdPrs);
  Hid.SetReportParser(0, &MousePrs);
```

K1NG7005, 2022-06-12:
> > And you need to cheat at games because you suck :-)
> 
> People cheat because they can. Simple as that. You can say they suck but is it really their fault for devs lack of skills in implementing such checks?

bro my 502 dont work do u have any solution?

Acerdiprofessione, 2022-09-28:
any solutions ?