felis/USB_Host_Shield_2.0 #638: USBHIDBootMouse Mouse a4tech bloody P81

fessnecro, 2021-05-18:
Can't see the Y axis. X axis is good. I try to add int8_t dZ and use it like Y axis, but when I repeats it to computer, mouse is verry lagging. X is good, but Y does not react to small movements

Lauszus, 2021-05-18:
Please share your code.

fessnecro, 2021-05-18:
// *** ALL MY Code is 100% free of cost and liability *** 

//-----------------------------------------------------------------------------------//
#include <avr/pgmspace.h>
#include <Usb.h>
#include <usbhub.h>
#include <avr/pgmspace.h>
#include <hidboot.h>

//----------------------------------------//
// Usb / HID mouse
USB Usb;
HIDBoot<USB_HID_PROTOCOL_MOUSE> Mouse(&Usb);

//-----------------------------------------------------------------------------------//
// Memory/Capture/Playback
#define MAX_DELTAS 419
#define XmitDelay 10   // ms Delay between transmitted deltas (tuned)
const bool InvertAxis          = false;
volatile bool CompressionMode  = false;

//----------------------------------------//
struct {
    uint8_t buttons;
    int8_t x;
    int8_t y;
    int8_t wheel;   /* Not yet implemented */
} mouseReport;
//----------------------------------------//
//-----------------------------------------------------------------------------------//

//-----------------------------------------------------------------------------------//
void ProcessDelta(MOUSEINFO* Info) {
    RepeateDelta(Info); // Always repeate (it's hard to know where the pointer will end up)
}

//----------------------------------------------------------------------------------//
void RepeateDelta(MOUSEINFO* Info) {
    mouseReport.buttons = Info->bmLeftButton | ((Info->bmRightButton)<<1) | ((Info->bmMiddleButton)<<2);
    mouseReport.x = (InvertAxis)? -Info->dX : Info->dX;
    mouseReport.y = (InvertAxis)? -Info->dZ : Info->dZ;
    mouseReport.wheel = 0; // Not yet implemented in HEX!
    Serial.write((uint8_t*)&mouseReport, 4);
    delay(XmitDelay);
}
//-----------------------------------------------------------------------------------//
void FreezeInErrorState() {
    while(true) {}
}
//-----------------------------------------------------------------------------------//

class MouseRptParser : public MouseReportParser {
protected:
    virtual void OnMouseMove        (MOUSEINFO* mi);
    virtual void OnLeftButtonUp     (MOUSEINFO* mi);
    virtual void OnLeftButtonDown   (MOUSEINFO* mi);
    virtual void OnRightButtonUp    (MOUSEINFO* mi);
    virtual void OnRightButtonDown  (MOUSEINFO* mi);
    virtual void OnMiddleButtonUp   (MOUSEINFO* mi);
    virtual void OnMiddleButtonDown (MOUSEINFO* mi);
};
void MouseRptParser::OnMouseMove(MOUSEINFO* mi)        {ProcessDelta(mi);}
void MouseRptParser::OnLeftButtonUp(MOUSEINFO* mi)     {ProcessDelta(mi);}
void MouseRptParser::OnLeftButtonDown(MOUSEINFO* mi)   {ProcessDelta(mi);}
void MouseRptParser::OnRightButtonUp(MOUSEINFO* mi)    {ProcessDelta(mi);}
void MouseRptParser::OnRightButtonDown(MOUSEINFO* mi)  {ProcessDelta(mi);}
void MouseRptParser::OnMiddleButtonUp(MOUSEINFO* mi)   {ProcessDelta(mi);}
void MouseRptParser::OnMiddleButtonDown(MOUSEINFO* mi) {ProcessDelta(mi);}
MouseRptParser Prs;

void setup() {
    Serial.begin(9600);
    // Init USB/Mouse
    if (Usb.Init() == -1) FreezeInErrorState();
    delay(200);
    Mouse.SetReportParser(0,(HIDReportParser*)&Prs);
}

//-----------------------------------------------------------------------------------//
void loop() {
    Usb.Task();
}