felis/USB_Host_Shield_2.0 #643: Is this a USB Communication Issue / Additional USB Tests?

StevenMartin1, 2021-06-17:
Hello, 

Apologies if this is the wrong place to be asking for support help. 

I've posted my issue in (hopefully) great detail here: https://forum.arduino.cc/t/usb-midi-communication-error/874461 but so far I can't seem to find anyone who is able to help me with troubleshooting.

If no one is able to help with my specific issue.... Are there any other tests outside of the board_qc that I can use to verify USB communication to tell if I have fried something on the USB host shield or not? 

Huge thanks - and if posting this here is inappropriate, feel free to delete it. I'm just trying to get some help with an issue that's over my head!

Steve

Lauszus, 2021-06-17:
I took a quick look at your post. From the output of the `board_qc` example the and from that you can tell that the USB host shield is working fine and the USB device is detected, so it's not really an issue with this library, but an issue with your particular use case.

@YuuichiAkagawa is the developer of the MIDI driver and might be able to help you out with MIDI related issues.

YuuichiAkagawa, 2021-06-17:
"Transfer error: 13" means that hrJERR 
- J-state instead of response
See page 36 of https://pdfserv.maximintegrated.com/en/an/AN3785.pdf

It doesn't look like a software problem.

StevenMartin1, 2021-06-17:
Yikes, so the above two comments (which I'm highly appreciative of) seem to rule out hardware and software 😅 are there any other tests I could run to try and get to the bottom of what's happened here?

YuuichiAkagawa, 2021-06-17:
Is it possible to use the USBH_MIDI_dump example to check if messages from that MIDI device can be received?

StevenMartin1, 2021-06-17:
I believe the device I'm connecting to needs a few commands sent before it will spit out the replies, so I combined the USBH_MIDI_dump code with my test sketch I've been using. 

If I run it without any debugging I get:
```VID:0582, PID:01D8```

If I run it with debugging I get: 

```
> Executing task: platformio device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbmodem14301  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
MIDI Init
USBH_MIDI version 000600
Addr:01
VID:0582 PID:01D8 #Conf:01

STEP1: MIDI Start
Interface descriptor:
 Inf#:          00
 Alt:           00
 EPs:           00
 IntCl:         FF
 IntSubcl:      FF
 Protocol:      00
-MIDI not found

Interface descriptor:
 Inf#:          01
 Alt:           00
 EPs:           00
 IntCl:         FF
 IntSubcl:      02
 Protocol:      02
-MIDI not found

Interface descriptor:
 Inf#:          01
 Alt:           01
 EPs:           01
 IntCl:         FF
 IntSubcl:      02
 Protocol:      02
-MIDI not found

Interface descriptor:
 Inf#:          02
 Alt:           00
 EPs:           00
 IntCl:         FF
 IntSubcl:      02
 Protocol:      01
-MIDI not found

Interface descriptor:
 Inf#:          02
 Alt:           01
 EPs:           01
 IntCl:         FF
 IntSubcl:      02
 Protocol:      01
-MIDI not found

Interface descriptor:
 Inf#:          03
 Alt:           00
 EPs:           02
 IntCl:         FF
 IntSubcl:      03
 Protocol:      00
-MIDI not found

Interface descriptor:
 Inf#:          03
 Alt:           01
 EPs:           02
 IntCl:         FF
 IntSubcl:      03
 Protocol:      00
-MIDI not found

STEP1: MIDI,NumEP:01
MIDI not found.
STEP2: Attempts vendor specific bulk device
Interface descriptor:
 Inf#:          00
 Alt:           00
 EPs:           00
 IntCl:         FF
 IntSubcl:      FF
 Protocol:      00
Interface descriptor:
 Inf#:          01
 Alt:           00
 EPs:           00
 IntCl:         FF
 IntSubcl:      02
 Protocol:      02
Interface descriptor:
 Inf#:          01
 Alt:           01
 EPs:           01
 IntCl:         FF
 IntSubcl:      02
 Protocol:      02
>Extracting endpoint
Endpoint descriptor:
 Length:        07
 Type:          05
 Address:       0D
 Attributes:    05
 MaxPktSize:    0168
 Poll Intrv:    01
-No valid EP found.
Interface descriptor:
 Inf#:          02
 Alt:           00
 EPs:           00
 IntCl:         FF
 IntSubcl:      02
 Protocol:      01
Interface descriptor:
 Inf#:          02
 Alt:           01
 EPs:           01
 IntCl:         FF
 IntSubcl:      02
 Protocol:      01
>Extracting endpoint
Endpoint descriptor:
 Length:        07
 Type:          05
 Address:       8E
 Attributes:    25
 MaxPktSize:    0168
 Poll Intrv:    01
-No valid EP found.
Interface descriptor:
 Inf#:          03
 Alt:           00
 EPs:           02
 IntCl:         FF
 IntSubcl:      03
 Protocol:      00
>Extracting endpoint
Endpoint descriptor:
 Length:        07
 Type:          05
 Address:       03
 Attributes:    02
 MaxPktSize:    0040
 Poll Intrv:    01
+valid EP found.
>Extracting endpoint
Endpoint descriptor:
 Length:        07
 Type:          05
 Address:       84
 Attributes:    02
 MaxPktSize:    0040
 Poll Intrv:    00
+valid EP found.
Interface descriptor:
 Inf#:          03
 Alt:           01
 EPs:           02
 IntCl:         FF
 IntSubcl:      03
 Protocol:      00

STEP2: Vendor,NumEP:03
Conf:01
EPin :84
EPout:03
Init done.
VID:0582, PID:01D8
inTransfer():04
SendSysE Length:        000F
 Total pktSize: 0005
 wptr:  04
SendSysE Length:        000F
 Total pktSize: 0005
 wptr:  04
SendSysE Length:        000F
 Total pktSize: 0005
 wptr:  04
inTransfer():04
inTransfer():0D
SendSysE Length:        0012
 Total pktSize: 0006
 wptr:  04
inTransfer():04
inTransfer():0D
inTransfer():04
inTransfer():0D
inTransfer():04
inTransfer():0D

```

StevenMartin1, 2021-06-17:
Okay, so if I run the dump without debug it just shows the above UNLESS.... If I plug the usb from the amp into my mac and run the amp makers software... then plug it back into the arduino and I get:

```
VID:0582, PID:01D8
000016C5: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 04 4F F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000016DB: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 40 07 01 5A F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000016F1: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 07 38 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001707: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 10 43 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
0000171D: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 61 07 01 38 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001733: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 10 2F F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001749: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 24 2F F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
0000175F: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 15 2A F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001775: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 2A 29 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
0000178B: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 1A 25 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000017A1: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 33 20 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000017B7: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 1E 21 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000017CC: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 37 1C F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000017E3: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 22 1D F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000017F8: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 38 1B F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
0000180F: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 23 1C F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
0000184E: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 37 1C F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001864: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 20 1F F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001879: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 35 1E F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001890: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 1A 25 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000018A5: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 2E 25 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000018BC: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 16 29 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000018D1: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 26 2D F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000018E8: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 12 2D F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
000018FD: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 22 31 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
[... cut ...]
0000196C: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 05 48 07 18 3B F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01
00001982: 20: 04 F0 41 00 04 00 00 00 04 33 12 60 04 00 06 5B 07 0D 32 F7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 35 00 00 00 00 00 00 00 00 00 00 00 3C 3F DA 03 00 00 35 00 00 00 01 00 FF 00 01

```

Which makes me wonder if it's just something with the 'handshake' process with the amp? In which case my existing code from previous projects should work fine. 

i think the above proves it all works and its a 'me' problem?

YuuichiAkagawa, 2021-06-17:
I can't comment on device-specific behavior, but it may be that this amp will respond with 13 (0x0D) if the handshake does not complete successfully.
```
inTransfer():04
inTransfer():0D
```
- 04 : NAK
- 0D(13) :hrJERR

StevenMartin1, 2021-06-17:
Is there anything in https://github.com/felis/USB_Host_Shield_2.0/issues/643#issuecomment-863301937 that looks out of place to you? I'm starting to suspect that the shield is able to receive data but cannot send it out.  

Is there an example sketch that would test for this?

Lauszus, 2021-06-18:
@StevenMartin1 it can differently send out data or it would not be able to detect the USB device in the first place.

See if you can find a Linux driver or another open source driver, as it sounds to be that you might need to send some manufacturer specific commands in order for it to start sending out data. I do not have much experience with MIDI devices, but some game controller requires that.

StevenMartin1, 2021-06-19:
So that sounds like the shield/arduino are functioning okay and wired correctly, and this is purely a code issue?

In this use case, yes, the amp needs to receive two sysex 'handshake' messages and then a welcome email third to set it into an editor mode. The amp hasn't been updated or anything, and these handshakes etc are included in previously working (and since unchanged) backups of my project. 

What a head scratcher! Everywhere I turn things seem to be 'OK' but yet nothing is working 🤔😅

YuuichiAkagawa, 2021-06-20:
@Lauszus can he use the testusbhostFAT example as a test for OutTransfer?

StevenMartin1, 2021-06-28:
Can the above be used as a test?
Another thought, is there a way to check in/out functionality with something like a USB memory stick?

mwisniewski, 2025-02-27:
Any progress for that issue? I am facing the same problem.