felis/USB_Host_Shield_2.0 #646: hidboot:  Poll() buffer overrun

tmk, 2021-06-24:
Value from MaxPacketSize of Endpoint Descriptor is not checked before using, which can be upto 64 legally in full-speed, or more with malicious or buggy implementation.

Lauszus, 2021-06-24:
Good catch. Wouldn't it be better just to set the size of `buf` dynamically i.e:

```C
uint8_t buf[read];
```

tmk, 2021-06-26:
Nice idea. We can save stack space with that, instead of declaring 64 bytes always.

I tested this change with examples/HID/USBHIDBootKbd and confirmed that it works as expected.

@Lauszus 
Could you make a commit for this small change? Or is that better I make a PR?

```diff --git a/hidboot.h b/hidboot.h
index 4076400..4fe83ff 100644
--- a/hidboot.h
+++ b/hidboot.h
@@ -582,14 +582,18 @@ uint8_t HIDBoot<BOOT_PROTOCOL>::Poll() {
 
                 // To-do: optimize manually, using the for loop only if needed.
                 for(int i = 0; i < epMUL(BOOT_PROTOCOL); i++) {
-                        const uint16_t const_buff_len = 16;
-                        uint8_t buf[const_buff_len];
+                        const uint16_t const_buff_len = 64;
 
                         USBTRACE3("(hidboot.h) i=", i, 0x81);
                         USBTRACE3("(hidboot.h) epInfo[epInterruptInIndex + i].epAddr=", epInfo[epInterruptInIndex + i].epAddr, 0x81);
                         USBTRACE3("(hidboot.h) epInfo[epInterruptInIndex + i].maxPktSize=", epInfo[epInterruptInIndex + i].maxPktSize, 0x81);
                         uint16_t read = (uint16_t)epInfo[epInterruptInIndex + i].maxPktSize;
 
+                        if (read > const_buff_len)
+                            read = const_buff_len;
+
+                        uint8_t buf[read];
+
                         rcode = pUsb->inTransfer(bAddress, epInfo[epInterruptInIndex + i].epAddr, &read, buf);
                         // SOME buggy dongles report extra keys (like sleep) using a 2 byte packet on the wrong endpoint.
                         // Since keyboard and mice must report at least 3 bytes, we ignore the extra data.
```

Lauszus, 2021-06-26:
It's better if you make a PR, then you get some credit for finding the bug as well :)

tmk, 2021-06-26:
Just made PR #647 

Thank you for your support on this project!

closed 2021-06-26 by tmk