felis/USB_Host_Shield_2.0 #649: BT is not recognized

sashaalehin, 2021-06-28:
My BT is not recognized, and I don't know why. I use library 1.5.0. My Bluetooth dongle is version 3. PS3BT with debugging reports:

PS3 Bluetooth Library Started
Unknown Device Connected - VID: 2563 PID: 0575
BTD Init Failed, error code: D1

sashaalehin, 2021-07-09:
Help me, please

Lauszus, 2021-07-26:
Please run the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino and post the output.

The code assumed that the USB device class is set to 0xe0 (USB_CLASS_WIRELESS_CTRL).

sashaalehin, 2021-09-18:
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		2563
Product ID:		0575
Revision ID:		0200
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0029
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			80
Max.pwr:		AF

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		03
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	100100012289000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0020
Polling interval:	08

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0020
Polling interval:	0A

Addr:1(0.0.1)