felis/USB_Host_Shield_2.0 #652: time to establish connection

mind2reality, 2021-07-14:
Hi,

First of all , thank you all very much for this fantastic Libray :)

It takes nearly exact 6 seconds to establish the connection to my PS3 navigation controller.
I also figured out, when I disconnect via PS Button and reconnect it takes also those 6 seconds.
Is there a way to speed that a up ?

best regards
MIND

xxxajk, 2021-07-15:
Probably not. USB has to do a lot of work, and then there's bluetooth protocol on top of that. Worse, this happens over SPI, so basically you have:

```
SPI -> USB -> Bluetooth
and on the way back
Bluetooth -> USB -> SPI
```
Generally, for everything, I've found SPI to be the bottleneck.
What you are seeing is the latency for each layer, which builds up.
To send out a single USB packet, the MCU has to talk over SPI, and send the data, which depending on the SPI speed, can take a few ms. Then tell the USB to send, which generally takes 2-3ms, and that's IF the device is ready, which isn't always the case.
MUCH happens on SPI, it has to tell the Bluetooth what's going on, and check if anything has happened since the last check. This multiplies the traffic on SPI and USB. This happens faster on other MCU with built-in USB and PC's because there's no SPI. In fact most operations don't even involve the CPU at all, and use DMA.
So, no, and 6 seconds, considering all the work that happens, is actually pretty good.

HisashiKato, 2021-07-17:
@xxxajk 
Really?

For example
This is a video of an Xbox One S controller via Bluetooth, repeatedly connecting and disconnecting.
It reconnecting in about 2 seconds.

(The Xbox One S controller requires to press the Xbox button for 5 to 10 seconds to turn it off.)

When the green LED on the Tank is flashing, it indicates that it is disconnected, and when it turns off, it indicates that the connection is complete.

The hardware speed is not fast, it is slow.
However, it will reconnect in about 2 seconds.

https://user-images.githubusercontent.com/68432957/126028296-e09609c2-db5d-4acb-be7f-6e094cedfe46.mp4

This Hardware
 Arduino Pro Mini 3.3V 8MHz (ATmega328P)
 Mini USB Host Shield (Connect to Arduino via SPI)
 TP-Link Bluetooth USB Adapter UB400 (Bluetooth 4.0)

This Software
 My modified USB Host Shield Libraly
 Sketches for Tank

@mind2reality 
Try editing settings.h to turn on debugging and display the log on the serial monitor.
And see where it's taking time.

(Translate Site used)

Lauszus, 2021-07-26:
@HisashiKato thanks for the response. Indeed 6 seconds sound like a lot.

@mind2reality as @HisashiKato also wrote. Please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0/#enable-debugging, so you can give us some more insight into what is taking so long time.