felis/USB_Host_Shield_2.0 #655: ACM::SetLineCoding failing on ESP32 because of wrong line coding size calculation

fionnInniti, 2021-08-10:
I am trying to communicate with a CDC mode USB device that is sending and receiving serial data. The SetLineCoding function hasn't been working for me so I cannot set it into the right mode, unless I modify the library.

[`uint8_t ACM::SetLineCoding(const LINE_CODING *dataptr)`](https://github.com/felis/USB_Host_Shield_2.0/blob/master/cdcacm.cpp#L320) calculates the length of [`LINE_CODING`](https://github.com/felis/USB_Host_Shield_2.0/blob/master/cdcacm.h#L110) using `sizeof (LINE_CODING)` and on the 32-bit ESP32, this evaluates to 8 bytes, rather than the 7 bytes that should be sent in the control transfer.

Hardcoding it as the following appears to solve the issue:
```
uint8_t lineCodingSize = 7;
uint8_t rv = ( pUsb->ctrlReq(bAddress, 0, bmREQ_CDCOUT, CDC_SET_LINE_CODING, 0x00, 0x00, bControlIface, lineCodingSize, lineCodingSize, (uint8_t*)dataptr, NULL));
```

xxxajk, 2021-08-14:
Actually, that's because this code wasn't expected to run on anything bigger than 8bit, and the struct needs to be packed.
There are a lot of areas in the code that will end up barfing because of this.
In a few weeks, I'll be adding ESP8266 and ESP32 to UHS3, which will do all of this properly, and without the silly need to poll in the loop() of your sketch ;-)

fionnInniti, 2021-08-16:
Thanks for staying on top of it! 

I'm calling it directly in a custom CDC class driver so your fix will also cover that?

xxxajk, 2021-08-17:
UHS3 allows customization as a feature, all without having to modify the library source by using in line library loading.
So yes, you can use the classes provided as a base class and build upon it.

fionnInniti, 2021-08-20:
Ok thanks. I will leave this open and if there's still an issue post-change, I will make a pull request

Ryzee119, 2021-08-20:
UHS3 has so many improvements. Good to see a few commits in the repo recently!

xxxajk, 2021-08-20:
Most of the recent work with UHS3 has been with EHCI controllers in PJRC's boards.  Pull requests are welcome...

UHS2 has it's place, with VERY small (AVR) boards, as USB is a heavy lift for these MCU's due to the limited RAM.

UHS3 is intended for use with MCU that provides at least 32K of RAM, but 64K or better is the lowest end recommendation to do anything complex.

ESP8266 and ESP32 support is in the plans, and I do have these, I just need to find the time, plus I'm trying to get the current parts stabilized.

FWIW, USB via SPI is going to be slow, it's a bottleneck. High speed USB is 480Mbit, which is why the best you will see on SPI is full speed, which is 12Mbit. The usual cautions to using very short connections, with balanced lengths applies. There's also the possibility to use USB over IP, which would give much better performance for these WiFi based chips.