felis/USB_Host_Shield_2.0 #656: alternative pins for ESP32

rodrigtti, 2021-08-10:
Hi guys , in my project i need to use other pins to SPI interface for ESP32 I tried to edit the avrpins.h as the example bellow with no luck

`MAKE_PIN(P19, 12); // MISO
MAKE_PIN(P23, 13); // MOSI
MAKE_PIN(P18, 14); // SCK`

reading the esp32 docs (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html) I found out that it has 2 general purpose SPI interface that we could use ,one is HSPI with pins are( 18,19,23) as same as it is proposed on this lib, and  VSPI (pins 14 ,12,13).
So my question is is possible to change HSPI to VSPI?

I also tried to change some config on SPI.H with no success

I changed this line :
`  SPIClass(uint8_t spi_bus=HSPI);`

to:

`  SPIClass(uint8_t spi_bus=VSPI);`

resulting the same message "OSC did not start"

DrJaymz, 2024-05-03:
I'm trying to do this as I don't have those pins available.  Both HSPI and VSPI can be mapped to other pins however for some reason in the overly complicated its done in this library its not that simple.

DrJaymz, 2025-08-20:
> did you ever end up figuring out how to do this?

Yes.

#define PIN_SPI_SCK 33 // SCK
#define PIN_SPI_MISO 36 // SCL
#define PIN_SPI_MOSI 5 // SDA
#define PIN_SPI_SS 32 // SS

I got it working on these pins.  I remember it being a real pig and I couldn't for the life of me work out why you couldn't just call a constructor with the pins.  I needed to get it working at the same time as ethernet and SPI TFT display.  In the end that works but the order of initialisation seemed to matter.

Looking at it, I have changed usbhost.h  My changes on the right:

[image]

[image]

So it looks like I tried to use the defines and then ended up having to chase it around with these pin defs.