felis/USB_Host_Shield_2.0 #657: USBHIDBootMouse many mouses does not work (wireless, glorious...)

EinfachMaxYT, 2021-08-11:
Hi :)

I already use some workarounds to use the different mice, unfortunately despite all this most of the mice do not work.

Almost no wireless mouse is supported, the glorious doesn't work, many other mice don't work either. I can get almost any Logitech mouse to work but that's where it stops.

Can anyone help me to fix wireless mouses and others?

Thank you! :)

FalkoBabbage, 2021-08-11:
I have not tried wireless mouses myself with USBHIDBootMouse. I think (not sure) that you should use BTHID (bluetooth HID), which apparently works differently.
Try this example:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/BTHID/BTHID.ino

EinfachMaxYT, 2021-08-11:
I use a Logitech G305 wireless, this mouse works with the normal libs i have. But most other mouses does not work. So thats maybe because the bluetooth? :D

Zak19930308, 2021-08-13:
Any solution? I have G Pro Wireless and dont work... also I have a G502 Hero and dont work too... my G502 dont turn on the lights

EinfachMaxYT, 2021-08-14:
> Any solution? I have G Pro Wireless and dont work... also I have a G502 Hero and dont work too... my G502 dont turn on the lights

G Pro wired works, G502 wired works. But wireless is not so easy. Contact me in Discord Pingu#0001 maybe i can help you, logitech is not a big deal normaly :)

xxxajk, 2021-08-14:
FWIW, if the mouse supports HID-BOOT, it should work in it's most basic mode.
If it does not support HID-BOOT, I've no idea :-)

EinfachMaxYT, 2021-08-15:
> Any solution? I have G Pro Wireless and dont work... also I have a G502 Hero and dont work too... my G502 dont turn on the lights

lol my discord not correct "Pinguu#0001"

Scalee, 2021-09-16:
> > Any solution? I have G Pro Wireless and dont work... also I have a G502 Hero and dont work too... my G502 dont turn on the lights
> 
> lol my discord not correct `Pinguu#0001`
> 
> I fixed Glarious Model o wireless and wired already

**Pinguu#0001 is not accepting friend requests. They'll have to add you to become friends.**
Maybe you can add me? :) Scale#2536

EinfachMaxYT, 2021-09-16:
> **Pinguu#1 is not accepting friend requests. They'll have to add you to become friends.**
> Maybe you can add me? :) Scale#2536

Ya sorry i added you. I got to much messages. :)

dylanrush, 2021-09-21:
Trying to get the G502 to work. My discord is dylanhrush#5952 . Tried sending you all friend requests on Discord.

dylanrush, 2021-09-22:
I got my wired G502 to work by using the HIDUniversal class instead of HIDBoot. I also had some trouble with the USB hub as order apparently matters.

So the magic incantation for me was:

```
HIDBoot < USB_HID_PROTOCOL_KEYBOARD | USB_HID_PROTOCOL_MOUSE > HidComposite(&Usb);
HIDUniversal                                    Hid(&Usb);

  HidComposite.SetReportParser(0, &KbdPrs);
  Hid.SetReportParser(0, &MousePrs);
```

.. where HidComposite represents my keyboard (a Filco Majestouch) and Hid represents my mouse.

ranchen421, 2022-02-25:
> Pinguu#0001

discord：Carrot#5428  I have a G502 Hero and dont work