felis/USB_Host_Shield_2.0 #658: Provide example of USB Hub code

cyberluke, 2021-08-14:
I have working USB Hid gaming keyboard containing uncommented line for USBHub and it is working. However when I connect this through ATEN KVM, which acts as a second USB hub and composite mouse keyboard device, your library is not able to enumerate any device and serial console output is empty. I try to find out the issue and spent several hours, but with no luck.

I found this code, could you please provide an example how to enumerate 2 USB hubs connected to each other + USB hid device connected to the 2nd hub? Your official example USB_desc have some code commented and it is difficult for me to understand it.

```#include <usbhub.h>

#include "pgmstrings.h"

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

USB     Usb;
//USBHub  Hub1(&Usb);
//USBHub  Hub2(&Usb);
//USBHub  Hub3(&Usb);
//USBHub  Hub4(&Usb);
//USBHub  Hub5(&Usb);
//USBHub  Hub6(&Usb);
//USBHub  Hub7(&Usb);

void PrintAllAddresses(UsbDevice *pdev)
{
  UsbDeviceAddress adr;
  adr.devAddress = pdev->address.devAddress;
  Serial.print("\r\nAddr:");
  Serial.print(adr.devAddress, HEX);
  Serial.print("(");
  Serial.print(adr.bmHub, HEX);
  Serial.print(".");
  Serial.print(adr.bmParent, HEX);
  Serial.print(".");
  Serial.print(adr.bmAddress, HEX);
  Serial.println(")");
}

void PrintAddress(uint8_t addr)
{
  UsbDeviceAddress adr;
  adr.devAddress = addr;
  Serial.print("\r\nADDR:\t");
  Serial.println(adr.devAddress, HEX);
  Serial.print("DEV:\t");
  Serial.println(adr.bmAddress, HEX);
  Serial.print("PRNT:\t");
  Serial.println(adr.bmParent, HEX);
  Serial.print("HUB:\t");
  Serial.println(adr.bmHub, HEX);
}

void setup()
{
  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
    Serial.println("OSC did not start.");

  delay( 200 );
}

uint8_t getdevdescr( uint8_t addr, uint8_t &num_conf );

void PrintDescriptors(uint8_t addr)
{
[... 254 lines cut ...]
void printunkdescr( uint8_t* descr_ptr )
{
  uint8_t length = *descr_ptr;
  uint8_t i;
  printProgStr(Unk_Header_str);
  printProgStr(Unk_Length_str);
  print_hex( *descr_ptr, 8 );
  printProgStr(Unk_Type_str);
  print_hex( *(descr_ptr + 1 ), 8 );
  printProgStr(Unk_Contents_str);
  descr_ptr += 2;
  for ( i = 0; i < length; i++ ) {
    print_hex( *descr_ptr, 8 );
    descr_ptr++;
  }
}

/* Print a string from Program Memory directly to save RAM */
void printProgStr(const char* str)
{
  char c;
  if (!str) return;
  while ((c = pgm_read_byte(str++)))
    Serial.print(c);
}

cyberluke, 2021-08-14:
USBHub class contains this header file and I don't know how to use it. I don't need polling. I just want to enumerate devices once during the boot and then get keyboard parser class on it. Also PrintHubStatus() is not implemented in C file, it contains only empty method stub, so I am not able to call that method at all.

```
public:
        USBHub(USB *p);

        uint8_t ClearHubFeature(uint8_t fid);
        uint8_t ClearPortFeature(uint8_t fid, uint8_t port, uint8_t sel = 0);
        uint8_t GetHubDescriptor(uint8_t index, uint16_t nbytes, uint8_t *dataptr);
        uint8_t GetHubStatus(uint16_t nbytes, uint8_t* dataptr);
        uint8_t GetPortStatus(uint8_t port, uint16_t nbytes, uint8_t* dataptr);
        uint8_t SetHubDescriptor(uint8_t port, uint16_t nbytes, uint8_t* dataptr);
        uint8_t SetHubFeature(uint8_t fid);
        uint8_t SetPortFeature(uint8_t fid, uint8_t port, uint8_t sel = 0);

        void PrintHubStatus();

        uint8_t Init(uint8_t parent, uint8_t port, bool lowspeed);
        uint8_t Release();
        uint8_t Poll();
        void ResetHubPort(uint8_t port);

        virtual uint8_t GetAddress() {
                return bAddress;
        };

        virtual bool DEVCLASSOK(uint8_t klass) {
                return (klass == 0x09);
        }

xxxajk, 2021-08-14:
Hubs are normally just dumb devices, and there's not much to do with them, except pass packets to/from devices up stream of the hub. What are you expecting?

cyberluke, 2021-08-15:
Did you check the code at all, please? This starts to feel stupid for me, but this method is not called anywhere: printhubdescr

So I want to print hub information in order to debug this stuff. Provide complete example of the above.

Next thing: there is open github issue where some hub id property (not sure how is it called exactly now) is not set correctly when connecting 2 hubs.

If already searched whole issues here, browsed your source code, tried to experiment. If it would work with Aten KVM and/or two hubs in a row, I would not need to post here as a developer. I did not expect this dumb answer here. I did expect some more professional help or advice. Thank you for understanding.

cyberluke, 2021-08-15:
1) Your official example USB_desc contains some methods to print hub information, but they are not used. There is zero information how to enumerate multiple hubs or how to use printhubdescr() method.

2) There is zero information how your USBHub class works, what is required to do in order to use one hub.

3) What is required to do to use two hubs

4) There is some issue about wrong hub id here and it is confirmed to be a bug: https://github.com/felis/USB_Host_Shield_2.0/issues/644

5) I already posted a specific code and asked specific questions. For example what is PrintHubStatus() doing in USBHub header file plus some other questions.

I did not like previous response: none of my questions were answered and only some question WHY have been asked.

Lauszus, 2021-08-15:
Here's the USB hub example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/hub_demo/hub_demo.ino

In order to use a USB hub you simply add the following to your code:

```C++
#include <usbhub.h>
USBHub Hub1(&Usb);
```

That's it. The library will take care of the rest. There should be no need for you to access the different function available in the USBHub instance.

MichaelPNolan, 2021-12-08:
I'm trying to figure out if uint8_t SetHubFeature(uint8_t fid); can be used to make interrupts function when there is some incoming packet so I can service the USB via interrupt driven code

xxxajk, 2021-12-09:
That's not going to be easy with this code base. For ISR-based code, use UHS3

chrisp250, 2022-02-06:
> Here's the USB hub example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/hub_demo/hub_demo.ino
>
> In order to use a USB hub you simply add the following to your code:
>
> ```c++ [...]

Hi,
Is there a way to get a USB object for a device on a specific hub port?
Thank you

xxxajk, 2022-02-07:
No, but USH30 can do that.

Lauszus, 2022-02-07:
Yes there is. Fx use `GetUsbDevicePtr` or `ForEachUsbDevice` available in the address poll (you can get that using `GetAddressPool`). Feel free to expand on the API and send a PR.