felis/USB_Host_Shield_2.0 #660: my keyboard not working

RomaNcy1, 2021-08-16:
First of all, I hope for your help, because I really need to connect this keyboard very much. 
Secondly, I bought an Arduino Uno, a Lenardo and a Mini, and I tried all of this and the keyboard did not work for me. when i use sketch hidkb.io Nothing happens when i use sketch hid multi kb and mouse see word start in serial monitor I see word start When I press any button on the keyboard, it says "start" 
my keyboard Is drevo blademaster with arm cortex 32bit mcu 
I bought all these parts to convert this keyboard to bluetooth 
ftdi 
hc05 bt with nr42 firmware 
ttl
Uno
Lenardo 
nano
Mini
usb host shield 
I hope you can help me please

RomaNcy1, 2021-08-17:
**EDIT** 

 **try-1
 enable debugging with arduino leonardo with USBHIDMultimediaKbd**

Start
HU Init
Addr:01
NC:01
Cnf:01
NumIface:02
SetIdle:00
SetIdle:01
HU configured

00 00 29 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 3B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 3B 3A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 3A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 3B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 29 3B 3A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

**try -2
enable debugging with arduino leonardo with USBHIDBootKbdAndMouse ---- but my arduino It stops or slows down and does not respond until it is restarted again**

Start
BM Init
01
HID_PROTOCOL_KEYBOARD AND MOUSE
bNumEP:03
Cnf:01
bIfaceNum:01
bNumIface:00

Interface:00
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
RPIPE rcode:00

Interface:01
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
RPIPE rcode:00
BM configured
M Butt Dn
dx=0 dy=70
04 00 46 29 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

**try -3
enable debugging with arduino uno with USBHID_desc**

Start
Count(10)
Report Size(01)
Input(00000010)
End Collection
Usage Page Gen Desktop Ctrls(01)Start
HU Init
Addr:01
NC:01
Cnf:01
NumIface:02
SetIdle:00
SetIdle:01
HU configured
0000: 05 01 09 06 A1 01 85 02 05 07 19 E0 29 E7 15 00 
0010: 25 01 75 01 95 08 81 02 95 01 75 08 81 03 95 05 
0020: 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 03 
0030: 95 3D 75 08 15 00 26 98 00 05 07 19 00 29 98 81 
0040: 00 C0 05 0C 09 01 A1 01 85 03 09 E9 09 EA 09 E2 
0050: 0A 92 01 0A 94 01 0A 23 02 0A 8A 01 0A 21 02 0A 
0060: 83 01 09 B6 09 CD 09 B5 09 B7 0A 2A 02 0A 24 02 
0070: 0A 25 02 15 00 25 01 95 10 75 01 81 02 C0 05 01 

Usage Page Gen Desktop Ctrls(01)
Usage Keypad
Collection Application
Report Id(02)
Usage Page Kbrd/Keypad(07)
Usage Min(E0)
Usage Max(E7)
Logical Min(00)
Logical Max(01)
Report Size(01)
Report Count(08)
Input(00000010)
Report Count(01)
Report Size(08)
Input(00000011)
Report Count(05)
Report Size(01)
Usage Page LEDs(08)
Usage Min(01)
Usage Max(05)
Output(00000010)
Report Count(01)
Report Size(03)
Output(00000011)
Report Count(3D)
Report Size(08)
Logical Min(00)
Logical Max(9800)
Usage Page Kbrd/Keypad(07)
Usage Min(00)
Usage Max(98)
Input(00000000)
End Collection
Usage Page Consumer(0C)
Usage Consumer Ctrl
Collection Application
Report Id(03)
Usage Vol Inc
Usage Vol Dec
Usage Mute
Usage AL Calc
Usage AL Loc Mach Brow
Usage AC Scroll Up
Usage AL E-mail Rdr
Usage AC Norm View
Usage AL Cons Ctrl Cfg
Usage Prev Track
Usage Play/Pause
Usage Next Track
Usage Stop
Usage AC Tile Horiz
Usage AC Scroll Down
Usage AC Scroll
Logical Min(00)
Logical Max(01)
Report Count(10)
Report Size(01)
Input(00000010)
End Collection
Usage Page Gen Desktop Ctrls(01)

tmk, 2021-08-18:
USBHIDMultimediaKbd seems to work as expected at least.

What model number is your keyboard?

And try my sketch to get a bit more detail of USB descriptors.
https://github.com/tmk/USB_Host_Shield_2.0/tree/upstream_add_desc_dump/examples/USB_desc_dump

RomaNcy1, 2021-08-18:
> USBHIDMultimediaKbd seems to work as expected at least.
> 
> What model number is your keyboard?
> 
> And try my sketch to get a bit more detail of USB descriptors.
> https://github.com/tmk/USB_Host_Shield_2.0/tree/upstream_add_desc_dump/examples/USB_desc_dump

thanks for reply
my keyboard Is **drevo blademaster te** with arm cortex 32bit mcu
he working but with enable debugging if disable debugging not working in serial monitor !!
i will try you sketch and i will give you more detail 
wait me

tmk, 2021-08-18:
USBHIDMultimediaKbd displays data **only when debug mode is eabled**.
This may not be useful as an example especially for first-timer. But it was written like that, anyway.

You can display data using `Serial.print()` without enabling debug.
```
diff --git a/examples/HID/USBHIDMultimediaKbd/USBHIDMultimediaKbd.ino b/examples/HID/USBHIDMultimediaKbd/USBHIDMultimediaKbd.ino
index 73f81e5..d3f69ef 100644
--- a/examples/HID/USBHIDMultimediaKbd/USBHIDMultimediaKbd.ino
+++ b/examples/HID/USBHIDMultimediaKbd/USBHIDMultimediaKbd.ino
@@ -31,10 +31,11 @@ bool HIDSelector::SelectInterface(uint8_t iface, uint8_t proto)
 void HIDSelector::ParseHIDData(USBHID *hid, uint8_t ep, bool is_rpt_id, uint8_t len, uint8_t *buf) {
 #if 1
   if (len && buf)  {
-    Notify(PSTR("\r\n"), 0x80);
+    Serial.print(F("\r\n"));
     for (uint8_t i = 0; i < len; i++) {
-      D_PrintHex<uint8_t > (buf[i], 0x80);
-      Notify(PSTR(" "), 0x80);
+      if (buf[i] < 16) Serial.print(F("0"));
+      Serial.print(buf[i], HEX);
+      Serial.print(F(" "));
     }
   }
 #endif
```

RomaNcy1, 2021-08-18:
> USBHIDMultimediaKbd seems to work as expected at least.
> 
> What model number is your keyboard?
> 
> And try my sketch to get a bit more detail of USB descriptors.
> https://github.com/tmk/USB_Host_Shield_2.0/tree/upstream_add_desc_dump/examples/USB_desc_dump

this is dump 

Start
Start
usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
usb_state: 90

//////////////////////////////////////////////////////////////////////
// USB_desc_dump
// Address: 01
// Lowspeed: 00

// Devicer dump:
12 01 00 02 00 00 00 08 2C 1A 1F B5 00 01 01 02
00 01

// Device:
bLength:		12
bDescriptorType:	01
bcdUSB:			0200
bDeviceClass:		00
bDeviceSubClass:	00
bDeviceProtocol:	00
bMaxPacketSize0:	08
idVendor:		1A2C
idProduct:		B51F
bcdDevice:		0100
iManufacturer:		01
iProduct:		02
iSerialNumber:		00
bNumConfigurations:	01

// String Zero: len: 04
// 04 03 09 04
// LangId: 0409

// iManufacturer: String1(0409): len: 14
// 14 03 44 00 52 00 45 00 56 00 4F 00 2E 00 49 00
// 6E 00 63 00
// DREVO.Inc

// iProduct: String2(0409): len: 26
// 26 03 42 00 6C 00 61 00 64 00 65 00 4D 00 61 00
// 73 00 74 00 65 00 72 00 20 00 54 00 45 00 20 00
// 38 00 37 00 4B 00
// BladeMaster TE 87K

// Config0 dump:	len: 003B
09 02 3B 00 02 01 00 A0 32 09 04 00 00 01 03 01
01 00 09 21 10 01 00 01 22 CF 00 07 05 82 03 40
00 01 09 04 01 00 01 03 01 01 00 09 21 10 01 00
01 22 40 00 07 05 81 03 40 00 01

// Config:
bLength:		09
bDescriptorType:	02
wTotalLength:		003B
bNumInterfaces:		02
bConfigurationValue:	01
iConfiguration:		00
bmAttributes:		A0
bMaxPower:		32

// Interface0.0:
bLength:		09
bDescriptorType:	04
bInterfaceNumber:	00
bAlternateSetting:	00
bNumEndpoints:		01
bInterfaceClass:	03
bInterfaceSubClass:	01
bInterfaceProtocol:	01
iInterface:		00

// HID: 
bLength:		09
bDescriptorType:	21
bcdHID:			0110
bCountryCode:		00
bNumDescriptors:	01
bDescrType:		22
wDescriptorLength:	00CF

// Report0 dump:	len: 00CF
05 01 09 06 A1 01 85 02 05 07 19 E0 29 E7 15 00
25 01 75 01 95 08 81 02 95 01 75 08 81 03 95 05
75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 03
95 3D 75 08 15 00 26 98 00 05 07 19 00 29 98 81
00 C0 05 0C 09 01 A1 01 85 03 09 E9 09 EA 09 E2
0A 92 01 0A 94 01 0A 23 02 0A 8A 01 0A 21 02 0A
83 01 09 B6 09 CD 09 B5 09 B7 0A 2A 02 0A 24 02
0A 25 02 15 00 25 01 95 10 75 01 81 02 C0 05 01
09 02 A1 01 85 04 09 01 A1 00 05 09 19 01 29 05
15 00 25 01 75 01 95 05 81 02 75 03 95 01 81 01
05 01 09 30 09 31 09 38 15 81 25 7F 75 08 95 03
81 06 C0 C0 06 00 FF 09 01 A1 01 85 05 75 08 95
07 15 00 26 FF 00 09 05 B1 02 09 06 91 02 C0

// Report:
Usage Page Gen Desktop Ctrls(01)
Usage Keypad
Collection Application
Report Id(02)
Usage Page Kbrd/Keypad(07)
Usage Min(E0)
Usage Max(E7)
Logical Min(00)
Logical Max(01)
Report Size(01)
Report Count(08)
Input(00000010)
Report Count(01)
Report Size(08)
Input(00000011)
Report Count(05)
Report Size(01)
Usage Page LEDs(08)
Usage Min(01)
Usage Max(05)
Output(00000010)
Report Count(01)
Report Size(03)
Output(00000011)
Report Count(3D)
Report Size(08)
Logical Min(00)
Logical Max(9800)
Usage Page Kbrd/Keypad(07)
Usage Min(00)
Usage Max(98)
Input(00000000)
End Collection
Usage Page Consumer(0C)
Usage Consumer Ctrl
Collection Application
Report Id(03)
Usage Vol Inc
Usage Vol Dec
Usage Mute
Usage AL Calc
Usage AL Loc Mach Brow
Usage AC Scroll Up
Usage AL E-mail Rdr
Usage AC Norm View
Usage AL Cons Ctrl Cfg
Usage Prev Track
Usage Play/Pause
Usage Next Track
Usage Stop
Usage AC Tile Horiz
Usage AC Scroll Down
Usage AC Scroll
Logical Min(00)
Logical Max(01)
Report Count(10)
Report Size(01)
Input(00000010)
End Collection
Usage Page Gen Desktop Ctrls(01)
Usage Mouse
Collection Application
Report Id(04)
Usage Pointer
Collection Physical
Usage Page Button(09)
Usage Min(01)
Usage Max(05)
Logical Min(00)
Logical Max(01)
Report Size(01)
Report Count(05)
Input(00000010)
Report Size(03)
Report Count(01)
Input(00000001)
Usage Page Gen Desktop Ctrls(01)
Usage X
Usage Y
Usage Wheel
Logical Min(81)
Logical Max(7F)
Report Size(08)
Report Count(03)
Input(00000110)
End Collection
End Collection
Usage Page Undef(00)
Usage
Collection Application
Report Id(05)
Report Size(08)
Report Count(07)
Logical Min(00)
Logical Max(FF00)
Usage
Feature(00000010)
Usage
Output(00000010)
End Collection

// Endpoint:
bLength:		07
bDescriptorType:	05
bEndpointAddress:	82
bmAttributes:		03
wMaxPacketSize:		0040
bInterval:		01

// Interface1.0:
bLength:		09
bDescriptorType:	04
bInterfaceNumber:	01
bAlternateSetting:	00
bNumEndpoints:		01
bInterfaceClass:	03
bInterfaceSubClass:	01
bInterfaceProtocol:	01
iInterface:		00

// HID: 
bLength:		09
bDescriptorType:	21
bcdHID:			0110
bCountryCode:		00
bNumDescriptors:	01
bDescrType:		22
wDescriptorLength:	0040

// Report0 dump:	len: 0040
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
75 01 95 08 81 02 95 01 75 08 81 03 95 05 75 01
05 08 19 01 29 05 91 02 95 01 75 03 91 03 95 3E
75 08 15 00 26 98 00 05 07 19 00 29 98 81 00 C0

// Report:
Usage Page Gen Desktop Ctrls(01)
Usage Keypad
Collection Application
Usage Page Kbrd/Keypad(07)
Usage Min(E0)
Usage Max(E7)
Logical Min(00)
Logical Max(01)
Report Size(01)
Report Count(08)
Input(00000010)
Report Count(01)
Report Size(08)
Input(00000011)
Report Count(05)
Report Size(01)
Usage Page LEDs(08)
Usage Min(01)
Usage Max(05)
Output(00000010)
Report Count(01)
Report Size(03)
Output(00000011)
Report Count(3E)
Report Size(08)
Logical Min(00)
Logical Max(9800)
Usage Page Kbrd/Keypad(07)
Usage Min(00)
Usage Max(98)
Input(00000000)
End Collection

// Endpoint:
bLength:		07
bDescriptorType:	05
bEndpointAddress:	81
bmAttributes:		03
wMaxPacketSize:		0040
bInterval:		01

// Parse data here: http://eleccelerator.com/usbdescreqparser/

RomaNcy1, 2021-08-18:
> USBHIDMultimediaKbd displays data **only when debug mode is eabled**.
> This may not be useful as an example especially for first-timer. But it was written like that, anyway.
>
> You can display data [...]

I can't add it to the sketch, can you make the file for me ready to add it on the arduino

touchgadget, 2021-08-18:
USBKBDMultimediaKbd appears to be working. The HID report data dumps are long probably because the keyboard is in NKRO mode.

If you have an N key rollover (NKRO) keyboard try switching to BIOS mode (also known as normal mode). NKRO mode uses unusual HID reports. I am not an expert at this but have run into NKRO keyboards that must be switched to BIOS mode to make them work with embedded USB host stacks. In BIOS mode, the keyboard uses 8 byte HID reports.

For example, look at this from the HID report desc. 61 * 8 bit HID report. 61 key rollover????

```
0x95, 0x3D,        //   Report Count (61)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0x98, 0x00,  //   Logical Maximum (152)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0x98,        //   Usage Maximum (0x98)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
```

These data dump lines are 64 bytes long. Not sure why not 61. Maybe the first byte is the modifier bitmap and second always zero. The bytes starting from the third byte look like USB HID key codes.
```
00 00 29 3B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 29 3B 3A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 29 3A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

```
ESC F2
ESC F2 F1
ESC F1
```

See this for USB HID key codes.

https://usb.org/sites/default/files/hut1_21.pdf

RomaNcy1, 2021-08-18:
> USBKBDMultimediaKbd appears to be working. The HID report data dumps are long probably because the keyboard is in NKRO mode.
>
> If you have an N key rollover (NKRO) keyboard try switching to BIOS mode [...]

First of all, I would like to thank you very much for this valuable information, but please excuse me, I am a very newbie... 
Secondly, I do not know how to turn on and off the Nkro mod 
Thirdly, I cannot use the code without activating the debugging, and I do not know how to add the code without activating the debugging 
Fourth, if possible, and I would be very thankful to you for your trouble to make a file for me ready to be added to the Arduino and make the keyboard work. Excuse me, I am very sorry for you

touchgadget, 2021-08-18:
> Secondly, I do not know how to turn on and off the Nkro mod
It is possible there is no option to do this on your keyboard. BIOS mode is also known as 6KRO mode.

An older Drevo keyboard supports 6KRO mode. https://www.amazon.com/DREVO-84-Key-Tenkeyless-Mechanical/dp/B01H6DHITE
I have no idea if this works on your keyboard. Google for "drevo 6kro" or "drevo bios" for more.

```
FN + 0()) = 6KRO, FN + -(_) = 24KRO, FN + F12 = Lock Windows. 
```

> Thirdly, I cannot use the code without activating the debugging, and I do not know how to add the code without activating the debugging

https://github.com/felis/USB_Host_Shield_2.0#enable-debugging

I think you have enabled debugging since the HID report hex dumps appear in your posting.

> Fourth, if possible, and I would be very thankful to you for your trouble to make a file for me ready to be added to the Arduino and make the keyboard work.
The code is working but the keyboard reports more than the usual data. You could leave the keyboard in NKRO mode but use only the first 8 bytes and ignore the rest. I have no idea how Bluetooth works.

RomaNcy1, 2021-08-18:
> An older Drevo keyboard supports 6KRO mode. https://www.amazon.com/DREVO-84-Key-Tenkeyless-Mechanical/dp/B01H6DHITE
> I have no idea if this works on your keyboard. Google for "drevo 6kro" or "drevo [...]

I have drevo blademaster te and I don't know how to turn off nkro mode 
Iam enable debugging but I want to turn off with sketch but When I disable it it doesn't give me any results in serial monitor One of the members told me how to solve this problem that I am adding some lines to the sketch, but I could not add it so I do not know how

tmk, 2021-08-18:
> I can't add it to the sketch, can you make the file for me ready to add it on the arduino

Check about `diff` and `patch` command on the net.
In summary, just remove lines marked '-' from original sketch and add lines marked '+'.

RomaNcy1, 2021-08-19:
> Check about `diff` and `patch` command on the net.
> In summary, just remove lines marked '-' from original sketch and add lines marked '+'.

yes it's working without enable debugging . thanks 
But as it is, it gives me the same result 
How is it handled now?? Please help me and I will be very thankful to you

touchgadget, 2021-08-19:
This may help decode the hex dumps.

https://wiki.osdev.org/USB_Human_Interface_Devices

RomaNcy1, 2021-08-19:
iam do not know how to decode the hex dumps 
i want any help to modify this sketch to workin without hex 

#include <hidcomposite.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

// Override HIDComposite to be able to select which interface we want to hook into
class HIDSelector : public HIDComposite
{
public:
    HIDSelector(USB *p) : HIDComposite(p) {};

protected:
    void ParseHIDData(USBHID *hid, uint8_t ep, bool is_rpt_id, uint8_t len, uint8_t *buf); // Called by the HIDComposite library
    bool SelectInterface(uint8_t iface, uint8_t proto);
};

// Return true for the interface we want to hook into
bool HIDSelector::SelectInterface(uint8_t iface, uint8_t proto)
{
  if (proto != 0)
    return true;

  return false;
}

// Will be called for all HID data received from the USB interface
void HIDSelector::ParseHIDData(USBHID *hid, uint8_t ep, bool is_rpt_id, uint8_t len, uint8_t *buf) {
#if 1
  if (len && buf)  {
   Serial.print(F("\r\n"));
    for (uint8_t i = 0; i < len; i++) {
    if (buf[i] < 16) Serial.print(F("0"));
    Serial.print(buf[i], HEX);
    Serial.print(F(" "));
    }
  }
#endif
}

USB     Usb;
//USBHub     Hub(&Usb);
HIDSelector    hidSelector(&Usb);

void setup()
{
  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
    Serial.println("OSC did not start.");

  // Set this to higher values to enable more debug information
  // minimum 0x00, maximum 0xff, default 0x80
  UsbDEBUGlvl = 0xff;

  delay( 200 );
}

void loop()
{
  Usb.Task();
}

RomaNcy1, 2021-08-19:
****any one here plase** **

RomaNcy1, 2021-08-20:
Please what do I do now?

RomaNcy1, 2021-08-20:
@tmk @touchgadget @felis @joebowbeer Have I reached the end and there is no other solution?? 
I hope there is a solution to this problem, I doubt that the solution is very simple, but I do not know it

RomaNcy1, 2021-08-20:
I changed HEX to DEC
But I got the same result on the serial monitor

xxxajk, 2021-08-20:
We can't know about hardware we don't own.

RomaNcy1, 2021-08-21:
> We can't know about hardware we don't own.

I am sorry and I understand what you mean, but try to help me. All you want to know is here. All the keyboard data is here. A full copy of its data was taken. Is it possible to reconsider the matter and see what I did and what I reached? 
I see that I am close to the end now, all the problem is that the data that is displayed is in hex format and I do not know what to solve!

touchgadget, 2021-08-21:
I am not clear what you want. If you want ASCII characters, try running this other example. I am not sure if it works with your NRKO keyboard but it works with a regular keyboard (6KRO).

https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino

RomaNcy1, 2021-08-21:
> I am not clear what you want. If you want ASCII characters, try running this other example. I am not sure if it works with your NRKO keyboard but it works with a regular keyboard (6KRO).
> 
> https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino

First, the keyboard does not give any data on this code /USBHIDBootKbd.ino 
Second, all I want is to operate the keyboard normally, meaning that it gives me the data of letters and numbers, not in hex format. 
I want to convert data from hex to ASCll characters
But I don't know how

RomaNcy1, 2021-08-21:
This is what I could come up with 
Now the keyboard writes letters and numbers, but the letters and numbers are out of place, meaning that when I press the letter z it gives me the letter t and so on 
Also, I don't know why only the letters do not appear without the very many 0 numbers that appear with the letters