felis/USB_Host_Shield_2.0 #665: Library not compatible with PiKVM Composite Device

akuker, 2021-09-26:
The [PiKVM](https://pikvm.org/) project uses a USB OTG to simulate a Composite Keyboard/Mouse device that also includes a mass storage device. I'm attempting to connect the PiKVM up to the USB Host Shield 2.0 library, but it appears that the USB Host Shield 2.0 library doesn't work with a Composite device that also has a mass storage device.

```
Composite KVM Device
Manufacturer: Pi-KVM
Config Number: 1
	Number of Interfaces: 3
	Interface Number: 0
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 01
	Interface Number: 1
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 00
	Interface Number: 2
		Name: usb-storage
		Alternate Number: 0
		Class: 08(stor.) 
		Sub Class: 06
```

When I trace through the code, it looks like UHS is able to process the keyboard and mouse endpoints. When it gets to the mass storage device, it reports a USB_DEV_CONFIG_ERROR_DEVICE_NOT_SUPPORTED error and doesn't finish initializing the device.

Below is the configuration of the PiKVM composite device:
```
Composite KVM Device
Manufacturer: Pi-KVM
Serial Number: CAFEBABE
Speed: 480Mb/s (high)
Bus:   1
Address:  16
USB Version:  2.00
Device Class: 00(>ifc )
Device Subclass: 00
Device Protocol: 00
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: 1d6b
Product Id: 0104
Revision Number:  1.00

Config Number: 1
	Number of Interfaces: 3
	Attributes: 80
	MaxPower Needed: 250mA

	Interface Number: 0
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 01
		Protocol: 01
		Number of Endpoints: 2

			Endpoint Address: 81
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 8
			Interval: 1ms

			Endpoint Address: 01
			Direction: out
			Attribute: 3
			Type: Int.
			Max Packet Size: 8
			Interval: 1ms

	Interface Number: 1
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 82
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 7
			Interval: 1ms

			Endpoint Address: 02
			Direction: out
			Attribute: 3
			Type: Int.
			Max Packet Size: 7
			Interval: 1ms

	Interface Number: 2
		Name: usb-storage
		Alternate Number: 0
		Class: 08(stor.) 
		Sub Class: 06
		Protocol: 50
		Number of Endpoints: 2

			Endpoint Address: 83
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 512
			Interval: 0ms

			Endpoint Address: 03
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 512
			Interval: 125us
``` 

Is there a way for the UHS library to continue processing the first two interfaces, even though the last interface isn't one that it understands?

I'm happy to try out any fixes or provide additional data.

xxxajk, 2021-09-28:
Use of a NON-AVR MCU (there are many out there, I prefer PJRC.com ARM offerings) and UHS3 *does* support this configuration and much more. 

UHS2, by design is not capable of this without merging stuff somehow, and is too much work. This  because UHS2 is device-centric, instead of interface-centric. 

Besides, USB is a heavy task for an AVR, and eats nearly all the RAM resources. The only way you can solve the problem on an AVR is to use a Mega2560 or 1280, and expand the memory on it using Xmem library. However this solution ends up costing more than an ARM MCU with enough RAM, and cost less than a Mega2560.

akuker, 2021-09-28:
> UHS2, by design is not capable of this without merging stuff somehow, and is too much work. This because UHS2 is device-centric, instead of interface-centric.

Is it possible for UHS2 to ignore the unexpected interface instead of failing to process the entire device?

xxxajk, 2021-09-28:
Nope. :-/

tmk, 2021-09-29:
How did you test and get the error?

I used this sketch successfully with some composite keyboards to dump HID report from any IN endpoint.
I think it ignores the Mass Storage interface.

USBHID_report_dump.ino
```
#include <hidcomposite.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

// Override HIDComposite to be able to select which interface we want to hook into
class HIDSelector : public HIDComposite
{
public:
    HIDSelector(USB *p) : HIDComposite(p) {};

protected:
    void ParseHIDData(USBHID *hid, uint8_t ep, bool is_rpt_id, uint8_t len, uint8_t *buf);
    bool SelectInterface(uint8_t iface, uint8_t proto);
};

bool HIDSelector::SelectInterface(uint8_t iface, uint8_t proto)
{
    return true;
}

// Will be called for all HID data received from the USB interface
void HIDSelector::ParseHIDData(USBHID *hid, uint8_t ep, bool is_rpt_id, uint8_t len, uint8_t *buf)
{
    if (len && buf)  {
        Serial.print(F("\r\nEp"));
        Serial.print(ep);
        Serial.print(F(": "));
        for (uint8_t i = 0; i < len; i++) {
            if (buf[i] < 16) Serial.print(F("0"));
            Serial.print(buf[i], HEX);
            Serial.print(F(" "));
        }
    }
}

USB         Usb;
USBHub      Hub(&Usb);
HIDSelector hidSelector(&Usb);

void setup()
{
    Serial.begin( 115200 );
#if !defined(__MIPSEL__)
    while (!Serial);
#endif
    Serial.println("Start");

    if (Usb.Init() == -1)
        Serial.println("OSC did not start.");

    delay( 200 );
}

static uint8_t prev_state = 0;

void loop()
{
    Usb.Task();

    // USB state
    if (prev_state != Usb.getUsbTaskState()) {
        prev_state = Usb.getUsbTaskState();
        Serial.print(F("usb_state: "));
        Serial.println(Usb.getUsbTaskState(), HEX);
    }
}
```

Chage this setting to enable debug prints.
```
diff --git a/settings.h b/settings.h
index 6b134df..1ff67c1 100644
--- a/settings.h
+++ b/settings.h
@@ -39,7 +39,7 @@ e-mail   :  support@circuitsathome.com
 ////////////////////////////////////////////////////////////////////////////////
 
 /* Set this to 1 to activate serial debugging */
-#define ENABLE_UHS_DEBUGGING 0
+#define ENABLE_UHS_DEBUGGING 1
 
 /* This can be used to select which serial port to use for debugging if
  * multiple serial ports are available.
```

xxxajk, 2021-09-29:
Composite devices are going to be nothing but trouble on UHS2. Save your sanity and time. Use something better.
Also, if you don't need the storage, or if you don't need the hid, you can ignore either on UHS3, and the parts that ARE recognized, will work. ;-)

akuker, 2021-10-02:
> How did you test and get the error?
> 
> I used this sketch successfully with some composite keyboards to dump HID report from any IN endpoint. I think it ignores the Mass Storage interface.

Holy cow! That worked! Its receiving the mouse and keyboard events from the pikvm! Thank you @tmk !

@xxxajk - I'll give UHS3 a try to see how well it works.

Thank you everyone!

dragon788, 2025-01-23:
You could also disable the MSD function of the PiKVM if it is yours to control.

https://docs.pikvm.org/msd/#disabling-mass-storage