felis/USB_Host_Shield_2.0 #671: Dynamic BT/USB Remote Instances

eMotimoJW, 2021-11-02:
Hi,

I am looking for any knowledge on the limitations of dynamically declaring your remote type. Currently working with PS4BT and I saw this old closed issue #78 . Are there any updates on this limitation and does it exist on all remote types? 

I have a goal of creating a Factory Pattern Interface to allow a user to select the remote type they would like to pair with but would be difficult to do without dynamic remote object creation/deletion. I've primarily used BT in the past but having an interface that would allow BT or USB would be the long term goal.

Any idea if this has been attempted before and/or if there are any existing limitations that would prevent this from working?

Thanks in Advance

eMotimoJW, 2021-11-02:
**Assuming a limitation on having to use static instances** at the expense of memory I have statically declared a PS4BT and a PS5BT and at runtime decide which one to pair. However, once a remote has been paired, connected, and later disconnected I attempt to pair the other controller. At this point the RemoteType.pair() does not seem to work and the second remote can never take over. If I restart the project the other remote pairs fine and can control the project. Any thoughts on what would limit me from switching which remote is paired without resetting the device?

Lauszus, 2021-11-04:
The only time I have experimented with dynamically creating BT instances is in the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/PS3Multi/PS3Multi.ino

Any change you have a PS3 controller you can test it with?

> Assuming a limitation on having to use static instances at the expense of memory I have statically declared a PS4BT and a PS5BT and at runtime decide which one to pair. However, once a remote has [...]

My guess is that something is not being reset in the BTHID or BTD classes. Unfortunately I do not have the time at the moment to dig into the issue.

Jaboop8, 2021-11-04:
Thanks @Lauszus  that example helped me resolve some memory faults I was hitting trying to implement a standard Factory Pattern.

As for the pairing issue I was seeing I think it has to do with the Parser() that is pointed to in the BTHID. When trying to pair the second remote type the BTHID still points to the Parser associated with the previous remote type. I ended up not needing to pursue this issue currently but thought it was worth noting if anyone else wants to play with it.

closed 2021-12-10 by Jaboop8