felis/USB_Host_Shield_2.0 #672: MSR90 Magstripe Card Reader not connected

Eddy12321, 2021-11-03:
Hello,

I am trying to get the MAX3421E to connect to a MSR90 magstripe card reader from amazon. However, the light on the card reader blinks continuously, signaling that it is not connected. I am running the USB_desc example so I expected it to work with no issues. I tested the MAX3421E with a Macally keyboard that I had laying around with the USB_desc example and the USBHIDBootKbd example and both worked perfectly. I also ran the Blink example successfully, so I am sure that my board works. The main difference between the keyboard and the card reader is that the keyboard is LOW speed and the card reader is FULL speed. I tested with other LOW and FULL speed devices and it seems that I can never connect to the FULL speed devices I have (gaming mouse, HP keyboard, and card reader), but have no problem connecting to LOW speed devices (mini mouse, Macally keyboard).

Here is the USB description of the card reader (from FTDI's USBView):
/////
Device Descriptor:
bcdUSB:             0x0110
bDeviceClass:         0x00
bDeviceSubClass:      0x00
bDeviceProtocol:      0x00
bMaxPacketSize0:      0x08 (8)
idVendor:           0xC216
idProduct:          0x0180
bcdDevice:          0x7B14
iManufacturer:        0x00
iProduct:             0x00
iSerialNumber:        0x00
bNumConfigurations:   0x01

ConnectionStatus: DeviceConnected
Current Config Value: 0x01
Device Bus Speed:     Full
Device Address:       0x11
Open Pipes:              1

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0008 (8)
bInterval:            0x01

Configuration Descriptor:
wTotalLength:       0x0034
bNumInterfaces:       0x02
bConfigurationValue:  0x01
iConfiguration:       0x00
bmAttributes:         0x80 (Bus Powered )
MaxPower:             0xFA (500 Ma)

Interface Descriptor:
bInterfaceNumber:     0x00
bAlternateSetting:    0x00
bNumEndpoints:        0x01
bInterfaceClass:      0x03 (HID)
bInterfaceSubClass:   0x01
bInterfaceProtocol:   0x01
iInterface:           0x00

HID Descriptor:
bcdHID:             0x0111
bCountryCode:         0x00
bNumDescriptors:      0x01
bDescriptorType:      0x22
wDescriptorLength:  0x002D

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0008 (8)
bInterval:            0x01

Interface Descriptor:
bInterfaceNumber:     0x01
bAlternateSetting:    0x00
bNumEndpoints:        0x00
bInterfaceClass:      0x03 (HID)
bInterfaceSubClass:   0x00
bInterfaceProtocol:   0x00
iInterface:           0x00

HID Descriptor:
bcdHID:             0x0111
bCountryCode:         0x00
bNumDescriptors:      0x01
bDescriptorType:      0x22
wDescriptorLength:  0x001A

//////

Here is what I get when I run board_qc with the card reader connected to the USB port:
///////
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
GPIO test passed.
PLL test. 100 chip resets will be performed
Resetting oscillator

Reset number 0 Time to stabilize - 333 cycles
Reset number 1 Time to stabilize - 333 cycles
Reset number 2 Time to stabilize - 332 cycles
Reset number 3 Time to stabilize - 332 cycles
Reset number 4 Time to stabilize - 332 cycles
Reset number 5 Time to stabilize - 332 cycles
Reset number 6 Time to stabilize - 332 cycles
Reset number 7 Time to stabilize - 332 cycles
Reset number 8 Time to stabilize - 333 cycles
Reset number 9 Time to stabilize - 332 cycles
Reset number 10 Time to stabilize - 332 cycles
Reset number 11 Time to stabilize - 332 cycles
Reset number 12 Time to stabilize - 333 cycles
Reset number 13 Time to stabilize - 333 cycles
Reset number 14 Time to stabilize - 333 cycles
Reset number 15 Time to stabilize - 333 cycles
Reset number 16 Time to stabilize - 333 cycles
Reset number 17 Time to stabilize - 332 cycles
Reset number 18 Time to stabilize - 332 cycles
Reset number 19 Time to stabilize - 332 cycles
Reset number 20 Time to stabilize - 332 cycles
Reset number 21 Time to stabilize - 332 cycles
Reset number 22 Time to stabilize - 332 cycles
Reset number 23 Time to stabilize - 332 cycles
Reset number 24 Time to stabilize - 332 cycles
Reset number 25 Time to stabilize - 332 cycles
Reset number 26 Time to stabilize - 333 cycles
Reset number 27 Time to stabilize - 333 cycles
Reset number 28 Time to stabilize - 333 cycles
Reset number 29 Time to stabilize - 333 cycles
Reset number 30 Time to stabilize - 333 cycles
Reset number 31 Time to stabilize - 333 cycles
Reset number 32 Time to stabilize - 332 cycles
Reset number 33 Time to stabilize - 332 cycles
Reset number 34 Time to stabilize - 332 cycles
Reset number 35 Time to stabilize - 332 cycles
Reset number 36 Time to stabilize - 332 cycles
Reset number 37 Time to stabilize - 332 cycles
Reset number 38 Time to stabilize - 332 cycles
Reset number 39 Time to stabilize - 332 cycles
Reset number 40 Time to stabilize - 332 cycles
Reset number 41 Time to stabilize - 333 cycles
Reset number 42 Time to stabilize - 333 cycles
Reset number 43 Time to stabilize - 333 cycles
Reset number 44 Time to stabilize - 333 cycles
Reset number 45 Time to stabilize - 333 cycles
Reset number 46 Time to stabilize - 332 cycles
Reset number 47 Time to stabilize - 332 cycles
Reset number 48 Time to stabilize - 332 cycles
Reset number 49 Time to stabilize - 332 cycles
Reset number 50 Time to stabilize - 332 cycles
Reset number 51 Time to stabilize - 332 cycles
Reset number 52 Time to stabilize - 332 cycles
Reset number 53 Time to stabilize - 332 cycles
Reset number 54 Time to stabilize - 332 cycles
Reset number 55 Time to stabilize - 332 cycles
Reset number 56 Time to stabilize - 333 cycles
Reset number 57 Time to stabilize - 333 cycles
Reset number 58 Time to stabilize - 333 cycles
Reset number 59 Time to stabilize - 333 cycles
Reset number 60 Time to stabilize - 333 cycles
Reset number 61 Time to stabilize - 332 cycles
Reset number 62 Time to stabilize - 332 cycles
Reset number 63 Time to stabilize - 332 cycles
Reset number 64 Time to stabilize - 333 cycles
Reset number 65 Time to stabilize - 332 cycles
Reset number 66 Time to stabilize - 332 cycles
Reset number 67 Time to stabilize - 332 cycles
Reset number 68 Time to stabilize - 332 cycles
Reset number 69 Time to stabilize - 332 cycles
Reset number 70 Time to stabilize - 332 cycles
Reset number 71 Time to stabilize - 333 cycles
Reset number 72 Time to stabilize - 333 cycles
Reset number 73 Time to stabilize - 333 cycles
Reset number 74 Time to stabilize - 333 cycles
Reset number 75 Time to stabilize - 333 cycles
Reset number 76 Time to stabilize - 332 cycles
Reset number 77 Time to stabilize - 332 cycles
Reset number 78 Time to stabilize - 332 cycles
Reset number 79 Time to stabilize - 333 cycles
Reset number 80 Time to stabilize - 332 cycles
Reset number 81 Time to stabilize - 332 cycles
Reset number 82 Time to stabilize - 332 cycles
Reset number 83 Time to stabilize - 332 cycles
Reset number 84 Time to stabilize - 332 cycles
Reset number 85 Time to stabilize - 333 cycles
Reset number 86 Time to stabilize - 333 cycles
Reset number 87 Time to stabilize - 333 cycles
Reset number 88 Time to stabilize - 333 cycles
Reset number 89 Time to stabilize - 333 cycles
Reset number 90 Time to stabilize - 333 cycles
Reset number 91 Time to stabilize - 332 cycles
Reset number 92 Time to stabilize - 332 cycles
Reset number 93 Time to stabilize - 332 cycles
Reset number 94 Time to stabilize - 332 cycles
Reset number 95 Time to stabilize - 332 cycles
Reset number 96 Time to stabilize - 332 cycles
Reset number 97 Time to stabilize - 332 cycles
Reset number 98 Time to stabilize - 332 cycles
Reset number 99 Time to stabilize - 332 cycles

Checking USB device communication.

Reset complete. Waiting for the first SOF...
USB state machine reached error state

/////

The card reader works like a keyboard so I expect it to work with USBHIDBootKbd, but it needs to be connected first, obviously. Any idea how to solve the FULL speed issue? Also, let me know if you need any further information. Thank you.

touchgadget, 2021-11-03:
> MaxPower: 0xFA (500 Ma)

The card reader device can draw as much as 500 mA so try powering the Arduino board through its barrel connector.

Eddy12321, 2021-11-03:
I measured with a current meter and it's really only drawing around 100 mA. However, I did try powering it with a desktop power supply (didn't have a dc jack supply), tried both the Vin pin at 9V and the 5V pin at 5V (the 5V pin is also connected to the usb v+), still no luck

Amnite, 2022-03-09:
I am actually doing the same.
MSR90 card reader, green led just blinks when connected to my XIAO(SAMD21).
I considered the MAX3421 but liked the idea of upgrading from the AtMega328 and AtTiny.
I also tried the better power supply to bo avail.
But some online documentation states blinking led is a card read. Idk. im on the same hunt though. So if I find something will report back.

Amnite, 2022-03-10:
A couple thoughts...
A) MSR90 - Internal  Hub?
B) Power MSR90 from 5v source?
C) https://github.com/gdsports/USB_Host_Library_SAMD
Imma give this library a shot when i get off work.
I know not all if any apply to you, but we both have the same problem, so i wouldn't imagine additional info could hurt.

Eddy12321, 2022-03-12:
@Amnite Did you have any luck with that library? I was never able to get the MAX3421E working with the MSR90 card readers. The error I was running into was "J-state instead of response". So, it seems that the MSR90 never takes control of the bus to respond. Wasn't sure how to fix this or why this error would be coming up. Searched the USB spec for info on this error, but didn't find much in the way of causes.

Amnite, 2022-03-14:
Well. I didn't have any luck with the other library. So I raged quit, and at $15 I decided to tear it open. Here is, maybe,  a ray of hope.

[image]
[image]

The IC on the back is labeled MH1601C. I have yet to dig up a datasheet on this; despite an hour or two of googling. The other IC is unlabeled but i have a sneaky suspicion it is a UART(?) to HID USB keyboard type thing. I've started tracing out the circuit, but would go alot faster if I could find ANYTHING  useful on either IC. I'm kinda noobish so may take a bit, as I will be learning as I disect it. So don't wanna dole out bogus info. If you find anything lemme know.
And btw. If it really is a UART then wiring to Arduino directly would be a cinch. The extra wires are what's making it fun. So if you beat me to something lemme know! And I will do the same. GL!