felis/USB_Host_Shield_2.0 #676: PS4 Controller unable to pair.

benjaminaarsen, 2021-11-28:
Hello, i have been trying to make this library work for a few hours now. But i don't seem to be able to get it working.
This is the output i get.

PS4 Bluetooth Library Started
Bluetooth Dongle Initialized
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:10
Please enable discovery of your device
Couldn't find HID device
Wait For Incoming Connection Request

The bluetooth dongle is flashing, and the host shield is working fine as well. Since the PS4USB works perfectly fine. It just somehow can't find the controller. When i use the dongle on my pc it is able to find the controller though... So communication with the dongle must be going wrong somewhere.

This is the dongle i used: https://www.aliexpress.com/item/33015867058.html?src=google&src=google&memo1=freelisting&albch=shopping&acnt=494-037-6276&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=6459980570&albag=76980386066&trgt=539263010115&crea=nl33015867058&netw=u&device=c&albpg=539263010115&albpd=nl33015867058&gclid=Cj0KCQiA7oyNBhDiARIsADtGRZYXVA8mmw_XzTdPg1HgpW-2Et6MjnLZF6As2fPoB3IW760sxFrsa4QaAj98EALw_wcB&gclsrc=aw.ds&aff_fcid=422fe2820d0f4c6cba418eace4f859a6-1638099471713-04895-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=422fe2820d0f4c6cba418eace4f859a6-1638099471713-04895-UneMJZVf&terminal_id=2f3d62296a4b48e4a0e29328d7ed07b4.

I would really appreciate if you could help me.

xxxajk, 2021-12-02:
If it requires firmware, then you are out of luck. UHS2 only works with "dumb" bluetooth dongles, and those may be hard to find.

benjaminaarsen, 2021-12-02:
How do i know whether the bluetooth dongle requires firmware or not? And do you perhaps know where to find the list of compatible bluetooth dongles? It was referenced on a few forums, but i can't seem to find it

Lauszus, 2021-12-02:
@xxxajk the dongle is recognized, so it does not require any firmware.
@benjaminaarsen there is an old list of dongles here: https://github.com/felis/USB_Host_Shield_2.0/wiki/Bluetooth-dongles. What I have done in the past is just to buy 10 cheap dongles from various sellers on eBay.

Please uncomment the `EXTRADEBUG` define here: https://github.com/felis/USB_Host_Shield_2.0/blob/master/BTD.cpp#L20. This will make it print out all devices it finds: https://github.com/felis/USB_Host_Shield_2.0/blob/master/BTD.cpp#L506-L554

Then we can at least see if it finds any devices.

benjaminaarsen, 2021-12-02:
Hi, I just tried this out. And it seems that the dongle does infact find the controller: 

PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:         07
Type:           05
Address:        83
Attributes:     03
MaxPktSize:     0010
Poll Intrv:     01
Endpoint descriptor:
Length:         07
Type:           05
Address:        01
Attributes:     02
MaxPktSize:     0040
Poll Intrv:     01
Endpoint descriptor:
Length:         07
Type:           05
Address:        81
Attributes:     02
MaxPktSize:     0040
Poll Intrv:     01
Bluetooth Dongle Initialized
Connection established
Connection established
Connection established
Connection established
Connection established
HCI Reset complete
Write class of device
Ignore HCI Event: 09
Local Bluetooth Address: 00:1A:7D:DA:71:10
HCI version: 09
Please enable discovery of your device
Unmanaged HCI Event: 2F, data: 01 4A 86 E5 11 AE A4 00 00 08 25 00 7B 49 D2 05 03 24 11 00 12 14 09 57 69 72 65 6C 65 73 73 20 43 6F 6E 74 72 6F 6C 6C 65 72 09 10 02 00 4C 05 CC 09 00 01 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Unmanaged HCI Event: 2F, data: 01 F1 4A 4B 6E 8C 9C 00 00 3C 04 08 05 B0 C1 09 10 01 00 E8 04 80 80 00 00 17 09 5B 54 56 5D 20 53 61 6D 73 75 6E 67 20 54 56 20 41 61 72 73 65 6E 09 03 0A 11 0C 11 0E 11 00 12 
01 05 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Unmanaged HCI Event: 2F, data: 01 4C A1 F8 FF DC E0 00 00 0C 02 5A 41 05 CF 0A 09 4D 69 20 39 54 20 50 72 6F 15 03 05 11 0A 11 0C 11 12 11 15 11 16 11 1F 11 2F 11 00 12 32 11 01 05 81 07 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Unmanaged HCI Event: 2F, data: 01 4C A2 33 0B 7A 04 00 00 24 04 28 39 2E BC 12 09 52 65 61 6C 74 65 6B 20 42 6C 75 65 74 6F 6F 74 68 0D 03 0A 11 0C 11 0E 11 15 11 16 11 00 12 01 05 81 07 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Couldn't find HID device
Wait For Incoming Connection Request
Ignore HCI Event: 1A

I decoded the data:
014A86E511AEA400000825007B49D20503241100121409576972656C65737320436F6E74726F6C6C6572091002004C05CC09000100000000000000
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000008A0100083C006904000100AA03010400690400010069040001006904000100690400010069040001006904000100690400010069040001006904000100690400010069040001006904000100690400010069040001006503000000000000000000000000000000000000000000000000000000

 of the first HCI event to a string and got this:

J†å®¤��%�{IÒ$�	Wireless Controller	�LÌ	��������?���������������������������������������������������������������������������������Š�<�i��ª�i��i��i��i��i��i��i��i��i��i��i��i��i��i��e���������������������������

so the controller is found, but i don't think the firmware recognizes it

Lauszus, 2021-12-02:
Seems like your dongle returns an "Extended Inquiry Result event" (0x2F). I can make a commit to a new branch if you can test it? :)

benjaminaarsen, 2021-12-02:
That would be awesome :)

Lauszus, 2021-12-02:
@benjaminaarsen see: https://github.com/felis/USB_Host_Shield_2.0/pull/677

benjaminaarsen, 2021-12-02:
It still doesn't work. It does show the 
Set inquiry mode to "Standard Inquiry Result event format"
after the commit: 

PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:         07  
Type:           05  
Address:        83  
Attributes:     03  
MaxPktSize:     0010
Poll Intrv:     01  
Endpoint descriptor:
Length:         07  
Type:           05  
Address:        01  
Attributes:     02  
MaxPktSize:     0040
Poll Intrv:     01  
Endpoint descriptor:
Length:         07  
Type:           05  
Address:        81  
Attributes:     02  
MaxPktSize:     0040
Poll Intrv:     01  
Bluetooth Dongle Initialized
Connection established
Connection established
Connection established
Connection established
Connection established
HCI Reset complete
Write class of device
Ignore HCI Event: 09
Local Bluetooth Address: 00:1A:7D:DA:71:10
HCI version: 09
Set inquiry mode to "Standard Inquiry Result event format"
Please enable discovery of your device
Unmanaged HCI Event: 2F, data: 01 4A 86 E5 11 AE A4 00 00 08 25 00 B5 DE CC 05 03 24 11 00 12 14 09 57 69 72 65 6C 65 73 73 20 43 6F 6E 74 72 6F 6C 6C 65 72 09 10 02 00 4C 05 CC 09 00 01 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Unmanaged HCI Event: 2F, data: 01 F1 4A 4B 6E 8C 9C 00 00 3C 04 08 11 F3 C0 09 10 01 00 E8 04 80 80 00 00 17 09 5B 54 56 5D 20 53 61 6D 73 75 6E 67 20 54 56 20 41 61 72 73 65 6E 09 03 0A 11 0C 11 0E 11 00 12 
01 05 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Unmanaged HCI Event: 2F, data: 01 4C A2 33 0B 7A 04 00 00 24 04 28 DB 71 B6 12 09 52 65 61 6C 74 65 6B 20 42 6C 75 65 74 6F 6F 74 68 0D 03 0A 11 0C 11 0E 11 15 11 16 11 00 12 01 05 81 07 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8A 01 00 08 3C 00 69 04 00 01 00 AA 03 01 04 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 69 04 00 01 00 65 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Couldn't find HID device
Wait For Incoming Connection Request

Lauszus, 2021-12-02:
Okay. Seems like your dongle doesn't respect the "HCI write inquiry mode". I'll update the code when I have time.

benjaminaarsen, 2021-12-02:
Thanks, I really appreciate the help.

Lauszus, 2021-12-19:
@benjaminaarsen can you try: https://github.com/felis/USB_Host_Shield_2.0/pull/681 and see if it fixes this issue.

benjaminaarsen, 2021-12-19:
Still not working but there is some improvement. It actually says it found the gamepad now. Although I don't think it gets recognized as a PS4 controller:

PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:         07  
Type:           05  
Address:        83  
Attributes:     03  
MaxPktSize:     0010
Poll Intrv:     01  
Endpoint descriptor:
Length:         07
Type:           05
Address:        01
Attributes:     02
MaxPktSize:     0040
Poll Intrv:     01
Endpoint descriptor:
Length:         07
Type:           05
Address:        81
Attributes:     02
MaxPktSize:     0040
Poll Intrv:     01
Bluetooth Dongle Initialized
Connection established
Connection established
Connection established
Connection established
Connection established
HCI Reset complete
Write class of device
Ignore HCI Event: 09
Local Bluetooth Address: 00:1A:7D:DA:71:10
HCI version: 09
Please enable discovery of your device
Number of responses: 1
Class of device: 00 25 08
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Number of responses: 1
Class of device: 28 04 24

Lauszus, 2021-12-19:
That event is a "Data Buffer Overflow event". Seems like it never manages to try to read the remote name of connect to the device before the the dongle overflows. Here's the full description:

> This event is used to indicate that the Controller’s data buffers have been
overflowed. This can occur if the Host has sent more packets than allowed.
The Link_Type parameter is used to indicate the type of data whose buffers
overflowed.

Lauszus, 2021-12-19:
Also for reference here is the output when it's working:

```
PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
Connection established
Connection established
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 34:C9:F0:8B:C9:3B
HCI version: 06
Please enable discovery of your device
Number of responses: 1
Class of device: 00 25 08
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Remote Name: Wireless Controller
PS4/PS5 controller is connecting
Connecting to HID device
Connection established
Connected to HID device
Unmanaged HCI Event: 11, data: 02 0B 00 A3 
Received Key Request
Bluetooth pin is set too: 0000
Pairing successful with HID device
Send HID Control Connection Request
Pairing was successful
Pairing was successful
Send HID Control Config Request
Pairing was successful
Pairing was successful
Set protocol mode: 00
Send HID Interrupt Connection Request
Send HID Interrupt Config Request
HID Channels Established
```

benjaminaarsen, 2021-12-19:
Yeah, mine doesn't getter further than the last part I sent. Is there anything you or I can do to fix the buffer overflow?

Lauszus, 2021-12-19:
Try to add:

```
if(rcode == hrNAK)
        return;
```

Just after:

```
uint8_t rcode = pUsb->inTransfer(bAddress, epInfo[ BTD_EVENT_PIPE ].epAddr, &length, hcibuf, pollInterval); // Input on endpoint 1
```

And post the output.

Lauszus, 2021-12-19:
FYI I've merged the PR into master, as it good to have and it also fixed a bug if multiple responses were returned in a normal inquiry result event.