felis/USB_Host_Shield_2.0 #678: Possible to use USB Host Shield as COM port to open terminal interface?

makrmark, 2021-12-04:
Hello,

I have a sensor that has a USB port that emulates a COM port on Windows so that it can be connected to via a terminal app and you can enter commands and receive response.

Is it possible to use USB Host Shield to provide a serial interface per the above - able to be accessed programatically via pySerial or similar?
e.g.:
https://opensource.com/article/20/5/usb-port-raspberry-pi-python

Thanks in advance,
Mark

xxxajk, 2021-12-04:
Short answer: yes, just be aware that you will lose hardware flow control.

makrmark, 2021-12-04:
> Short answer: yes, just be aware that you will lose hardware flow control.

Is there a long answer? :-)

The documentation doesn't seem to mention this scenario so I'm not sure where to start with an implementation.

Thanks.

touchgadget, 2021-12-05:
There are 3 USB serial/COM options. You will have to determine which one applies to your device.

See examples/acm, examples/ftdi, and examples/pl2303.

makrmark, 2021-12-05:
Thanks @touchgadget - I'm considering RaspberriPi Xero also but looking for lowest-power solution possible as this device will be away from power long-term (and I'd rather not build a solar solution for a number of reasons).